ktemkin/gnuradio-for-mac-without-macports #3: Perform a Runtime Check that XQuartz and Python2.7 are Installed

cfriedt, 2017-02-21:
GNURadio 3.7.10.1 depends on Python2.7 and the Mac OS X release depends on XQuartz.

The script 'run-grc' (or possibly just 'gnuradio-companion') should check for those dependencies and prompt the user to install them if they are missing.

dholl, 2017-05-27:
To test for XQuartz, execute ```open -a XQuartz``` and then test $?

```sh
open -a XQuartz
if [ $? -ne 0 ]; then
    printf 'XQuartz is not installed.\n' 1>&2
    exit 1
fi
```

Hmm, the downside to ```open -a XQuartz``` is that it causes XQuartz to launch an xterm window if XQuartz was not already running.  (This is a pet peeve of mine...)

Ahh, here's another way:
```sh
if [ "x${DISPLAY:-empty}" = "xempty" ]; then
    printf 'XQuartz is not installed.\n' 1>&2
    exit 1
fi
```

```$DISPLAY``` is set if XQuartz is available even if it isn't running yet as explained here:
https://www.xquartz.org/FAQs.html#dont-like-launchd-automatically-setting-up-display-for-you

dholl, 2017-05-28:
Ahh, here's another idea for testing the X server, run a utility that comes with X, such as xset:
```sh
if ! command -v xset >/dev/null 2>&1 || ! xset q >/dev/null 2>&1 ; then
    printf 'XQuartz is not available.\n' 1>&2
    exit 1
fi
```

Compared to just testing for non-empty ```$DISPLAY```, running ```xset``` will confirm that ```$DISPLAY``` actually points to a working X server's socket.

The ```command -v xset >/dev/null 2>&1``` first tests if the shell can find ```xset``` in $PATH.  (xset is another old utility that comes with just about every X server...). If ```xset``` isn't in the path, then we assume there's no X server.  Otherwise, then we try running ```xset q``` which queries the X server for some basic user preferences (keyboard repeat rate, blanking timeout, etc...) but we're only interested in the exit status to confirm if xset could contact the server or not.

Note that ```command``` and ```printf``` are shell builtin's, so no extra processes are spawned in these tests, except for the single invocation of ```xset```.
```
zero(ttys001):~> command -V command
command is a shell builtin
zero(ttys001):~> command -V printf 
printf is a shell builtin
```

dholl, 2017-05-28:
Regarding a test for python 2.7, how about something like:
```sh
if ! command -v python2.7 >/dev/null 2>&1 ; then
    printf 'Python 2.7 is not available.\n' 1>&2
    exit 1
fi
# Do we really need to verify that sys.version_info is 2.7 if the binary is called "python2.7"?
if [ "x2.7" != "x$(python2.7 -c 'import sys ; sys.stdout.write(".".join([str(n) for n in sys.version_info[:2]])+"\n")')" ]; then
    printf 'Python 2.7 is not available.\n' 1>&2
    exit 1
fi
```

or I could tweak the above to allow fallback to ```python``` if ```python2.7``` isn't found but just ```export VERSIONER_PYTHON_VERSION="2.7"``` and test sys.version_info again.

cfriedt, 2017-05-28:
There are about a million ways to do this, but the simplest is just to check for the existence of /Applications/Utilities/XQuartz.app

cfriedt, 2017-05-28:
Python is similar.

dholl, 2017-05-31:
> There are about a million ways to do this, but the simplest is just to check for the existence of /Applications/Utilities/XQuartz.app

Then
```sh
if ! test -d '/Applications/Utilities/XQuartz.app' ; then
    printf 'XQuartz.app is missing.\n' 1>&2
    exit 1
fi
```

But that still wouldn't indicate whether XQuartz is actually working.  :)  So far, I think my personal favorite for testing X presence and functionality is with ```xset```
```sh
if ! command -v xset >/dev/null 2>&1 || ! xset q >/dev/null 2>&1 ; then
    printf 'XQuartz is not available.\n' 1>&2
    exit 1
fi
```

But its your choice.  :)

cfriedt, 2017-07-15:
Going to add simple checks for Python2.7 and XQuartz. It's low-hanging fruit.

cfriedt, 2017-07-16:
Fixed in PR #53

closed 2017-07-16 by cfriedt