ktemkin/gnuradio-for-mac-without-macports #5: Splash Screen While GNURadio is Loading

cfriedt, 2017-02-21:
Would be nice to give an indication to the user that GNURadio (conoanion) has started. Bonus points for a progress bar.

cfriedt, 2017-05-27:
https://www.youtube.com/watch?v=wICS4RNQFBk
https://stackoverflow.com/questions/685062/how-to-show-a-splash-screen-during-launch-of-a-cocoa-app

dholl, 2017-05-27:
Heh, here's a 0-point solution that could be thrown into the immediate run-grc.sh:
```sh
xmessage -bd white -fg black -bg white -center -buttons '' -timeout 3 -title 'GNURadio' 'Launching GNURadio Companion ...' &
```

Caveats:  No progress bar, no buttons, just launches for a fixed duration such as 3 seconds in this case, and it is slightly ugly --- but perhaps the ugly can be reduced with -xrm arguments to override X widgets such as hiding the primitive scrollbar and changing the font.

Upside:  It comes with X11.  :)

dholl, 2017-05-28:
Hmm, I wonder if we wouldn't need any splash screen if we get gnuradio-companion running with a Mac-native pygtk that doesn't rely upon X11/XQuartz.  (much of the start-up delay on my system appears to just be waiting for XQuartz to run, but if XQuartz is already running, then clicking the GNURadio,app in Finder on my system only takes a second or two... --- but this is a late 2016 MBP with solid state drive, and the directory is likely already cached in ram...)

dholl, 2017-05-28:
Ah, here we go:
http://code.activestate.com/recipes/577919-splash-screen-gtk/

So assuming we get a Mac-native pygtk going to get around the XQuartz startup delay, if gnuradio-companion's own startup time is still too long, then we can try this gtk splash screen.

dholl, 2017-06-01:
Hmm, taking an idea from #3, how about this for ```run-grc.sh```:
```sh
#!/bin/sh

. '@INSTALL_DIR@/usr/bin/grenv.sh'
# TODO: This @INSTALL_DIR@ should really be replaced with something based
# on $0 to permit reloading GNURadio.app to other directories beside /Applications

if command -v xset >/dev/null 2>&1 ; then
	# If xset is available, then we'll use that to silently launch the X server.
	# While we wait, use osascript to tell the user what we're up to.
	osascript -e 'display notification "Starting X server..." with title "GNU Radio Companion"' 
	if xset q >/dev/null 2>&1 ; then
		osascript -e 'display notification "Starting GNU Radio Companion..." with title "GNU Radio Companion"'
	else
		osascript -e 'display notification "Unable to verify running X server.  Will attempt anyway..." with title "GNU Radio Companion"'
	fi
else
	osascript -e 'display notification "Starting GNU Radio Companion..." with title "GNU Radio Companion"'
fi

show_fail_message () {
	osascript -e 'display notification "Failed to start GNU Radio Companion." with title "GNU Radio Companion"' || true
}

# If the exec fails for some reason, then our EXIT handler will get invoked.
trap show_fail_message EXIT

exec gnuradio-companion "${@}"
```

On my laptop, if X isn't already running, then the longest delay for me it just waiting for X.  This script will launch X if it wasn't already running (by running ```xset q```).  When ```xset``` returns, then X must already be running (or just finished starting), and then ```gnuradio-companion``` should load must faster.

Give it a try and tell me what you think.  :)

It isn't a progress bar, but it should very quickly inform the user what we're up to...