ktemkin/gnuradio-for-mac-without-macports #9: Add OOTM Steps to README.md

cfriedt, 2017-02-22:
Add all of the steps required to build [out-of-tree modules](http://gnuradio.org/redmine/projects/gnuradio/wiki/OutOfTreeModules) to [README.md](https://github.com/cfriedt/gnuradio-for-mac-without-macports/blob/master/README.md). This will likely require splitting off part of [run-grc](https://github.com/cfriedt/gnuradio-for-mac-without-macports/blob/master/build.sh#L1290) into a sourceable environment (so that PYTHONPATH, DYLD_LIBRARY_PATH, etc, are extended).

Chicago1, 2018-06-29:
How do I do this?

cfriedt, 2018-07-01:
The key is to realize that most of GNURadio.app just relies on some external environment variables to work, for the most part.

Currently, most of those are defined in
- /Applications/GNURadio.app/Contents/MacOS/usr/bin/run-grc, and
- /Applications/GNURadio.app/Contents/MacOS/usr/bin/grenv.sh

You can view those files with "less /Applications/GNURadio.app/Contents/MacOS/usr/bin/<file>" in your terminal.

Then, I would suggest you follow the [out-of-tree-modules guide](http://gnuradio.org/redmine/projects/gnuradio/wiki/OutOfTreeModules) and include any additional steps you needed to do to get it working with GNURadio.app .

Thanks!

Chicago1, 2018-07-01:
> You can view those files with "less /Applications/GNURadio.app/Contents/MacOS/usr/bin/" in your terminal.

Thank you, I will try!

Chicago1, 2018-07-02:
I am trying to install http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver/, and the dev says to run "export GRC_BLOCKS_PATH=." but I cannot find this file in the application. Do you know where it is?

cfriedt, 2018-07-11:
Probably something like

/Applications/GNURadio.app/share/gnuradio/grc/blocks

See https://github.com/BastilleResearch/gr-lora/wiki/Installation