ktemkin/gnuradio-for-mac-without-macports #16: Fix PyQt4 Issues

cfriedt, 2017-03-11:
Seems that PyQt4 is not properly installed. Also, possibly that it is not in the `PYTHONPATH`

```
Executing: /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python -u /Users/cfriedt/Desktop/top_block.py

Traceback (most recent call last):
  File "/Users/cfriedt/Desktop/top_block.py", line 24, in <module>
    from gnuradio import qtgui
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/gnuradio/qtgui/__init__.py", line 37, in <module>
    from range import Range, RangeWidget
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/gnuradio/qtgui/range.py", line 24, in <module>
    from PyQt4 import Qt, QtCore, QtGui
ImportError: cannot import name QtGui
```

cfriedt, 2017-03-12:
I: Extracting PyQt4_gpl_x11-4.12.tar.gz to PyQt4_gpl_x11-4.12
    Determining the layout of your Qt installation...
    This is the GPL version of PyQt 4.12 (licensed under the GNU General Public
    License) for Python 2.7.10 on darwin.
    Found the license file pyqt-gpl.sip.
    Checking to see if the QtGui module should be built...
    Checking to see if the QtHelp module should be built...
    Checking to see if the QtMultimedia module should be built...
    Checking to see if the QtNetwork module should be built...
    Checking to see if the QtDBus module should be built...
    Checking to see if the QtDeclarative module should be built...
    Checking to see if the QtScript module should be built...
    Checking to see if the QtScriptTools module should be built...
    Checking to see if the QtXml module should be built...
    Checking to see if the QtOpenGL module should be built...
    Checking to see if the QtSql module should be built...
    Checking to see if the QtSvg module should be built...
    Checking to see if the QtTest module should be built...
    Checking to see if the QtWebKit module should be built...
    Checking to see if the QtXmlPatterns module should be built...
    Checking to see if the phonon module should be built...
    Checking to see if the QtAssistant module should be built...
    Checking to see if the QtDesigner module should be built...
    Checking to see if the QAxContainer module should be built...
    Checking to see if the dbus support module should be built...
    DBus v1 does not seem to be installed.
    Qt v4.8.7 free edition is being used.
    SIP 4.19.1 is being used.
    The Qt header files are in
    /Applications/GNURadio.app/Contents/MacOS/usr/include.
    The shared Qt libraries are in
    /Applications/GNURadio.app/Contents/MacOS/usr/lib.
    The Qt binaries are in /Applications/GNURadio.app/Contents/MacOS/usr/bin.
    The Qt mkspecs directory is in /Applications/GNURadio.app/Contents/MacOS/usr.
    These PyQt modules will be built: QtCore, QtGui, QtHelp, QtNetwork, QtXml,
    QtSql, QtSvg, QtTest, QtDesigner.
    The PyQt Python package will be installed in
    /Applications/GNURadio.app/Contents/MacOS/usr/lib/python2.7/site-packages.
    PyQt is being built with generated docstrings.
    PyQt is being built with 'protected' redefined as 'public'.
    The Designer plugin will be installed in
    /Applications/GNURadio.app/Contents/MacOS/usr/plugins/designer.
    The PyQt .sip files will be installed in
    /Applications/GNURadio.app/Contents/MacOS/usr/share/sip.
    pyuic4, pyrcc4 and pylupdate4 will be installed in
    /Applications/GNURadio.app/Contents/MacOS/usr/bin.
    Embedding sip flags...
    Generating the C++ source for the QtCore module...
    Creating the Makefile for the QtCore module...
    Generating the C++ source for the QtGui module...
    Creating the Makefile for the QtGui module...
    Generating the C++ source for the QtHelp module...
    Creating the Makefile for the QtHelp module...
    Generating the C++ source for the QtNetwork module...
    Creating the Makefile for the QtNetwork module...
    Generating the C++ source for the QtSql module...
    Creating the Makefile for the QtSql module...
    Generating the C++ source for the QtSvg module...
    Creating the Makefile for the QtSvg module...
    Generating the C++ source for the QtTest module...
    Creating the Makefile for the QtTest module...
    Generating the C++ source for the QtXml module...
    Creating the Makefile for the QtXml module...
    Generating the C++ source for the QtDesigner module...
    Creating the Makefile for the QtDesigner module...
    Generating the C++ source for the Qt module...
    Creating the Makefile for the Qt module...
    Creating QScintilla API file...
    Creating top level Makefile...
    Creating QPy support library for QtCore Makefile...
    Creating QPy support library for QtGui Makefile...
    Creating QPy support library for QtDesigner Makefile...
    Creating QPy support libraries Makefile...
    Creating pyuic4 wrapper...
    Creating pyuic4 Makefile...
    Creating pylupdate4 Makefile...
    Creating pyrcc4 Makefile...
    Creating Qt Designer plugin Makefile...
    Creating pyqtconfig.py...

cfriedt, 2017-03-12:
```
g++ -c -pipe -fPIC -arch x86_64 -O2 -Wall -W -DNDEBUG -DSIP_PROTECTED_IS_PUBLIC -Dprotected=public -DQT_NO_DEBUG -DQT_GUI_LIB -DQT_CORE_LIB -I. -I/Users/jdoe/Desktop/gnuradio-for-mac-without-macports/tmp/PyQt4_gpl_x11-4.12/qpy/QtGui -I/Applications/GNURadio.app/Contents/MacOS/usr/include -I/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7 -I/Applications/GNURadio.app/Contents/MacOS/usr/mkspecs/default -I/Applications/GNURadio.app/Contents/MacOS/usr/include/QtGui -I/Applications/GNURadio.app/Contents/MacOS/usr/include/QtCore -o sipQtGuicmodule.o sipQtGuicmodule.cpp
/Users/jdoe/Desktop/gnuradio-for-mac-without-macports/tmp/PyQt4_gpl_x11-4.12/QtGui/sipQtGuicmodule.cpp:9907:13: error: 
      use of undeclared identifier 'qRegisterDraggedTypes'
            qRegisterDraggedTypes(*a0);
            ^
/Users/jdoe/Desktop/gnuradio-for-mac-without-macports/tmp/PyQt4_gpl_x11-4.12/sip/QtGui/qapplication.sip:666:10: error: 
      use of undeclared identifier 'sipName_QX11EmbedContainer'
        {sipName_QX11EmbedContainer, &sipType_QX11EmbedContainer, -1, 74},
         ^
/Users/jdoe/Desktop/gnuradio-for-mac-without-macports/tmp/PyQt4_gpl_x11-4.12/sip/QtGui/qapplication.sip:666:39: error: 
      use of undeclared identifier 'sipType_QX11EmbedContainer'
        {sipName_QX11EmbedContainer, &sipType_QX11EmbedContainer, -1, 74},
                                      ^
/Users/jdoe/Desktop/gnuradio-for-mac-without-macports/tmp/PyQt4_gpl_x11-4.12/sip/QtGui/qapplication.sip:676:10: error: 
      use of undeclared identifier 'sipName_QX11EmbedWidget'
        {sipName_QX11EmbedWidget, &sipType_QX11EmbedWidget, -1, 80},
         ^
/Users/jdoe/Desktop/gnuradio-for-mac-without-macports/tmp/PyQt4_gpl_x11-4.12/sip/QtGui/qapplication.sip:676:36: error: 
      use of undeclared identifier 'sipType_QX11EmbedWidget'
        {sipName_QX11EmbedWidget, &sipType_QX11EmbedWidget, -1, 80},
                                   ^
5 errors generated.
make[1]: *** [sipQtGuicmodule.o] Error 1
make: *** [all] Error 2
```

cfriedt, 2017-04-23:
I believe PyQt4 issues are resolved. However, Qt itself has some issues remaining. Will add more details shortly.

cfriedt, 2017-04-28:
```
Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libQtGui.4.dylib              	0x0000000110117534 QWidget::inputMethodEvent(QInputMethodEvent*) + 4
1   libQtGui.4.dylib              	0x00000001103220e7 QFont::QFont(QFont const&, QPaintDevice*) + 55
2   libQtGui.4.dylib              	0x000000011010509b QWidgetPrivate::init(QWidget*, QFlags<Qt::WindowType>) + 971
3   libQtGui.4.dylib              	0x0000000110105267 QWidget::QWidget(QWidgetPrivate&, QWidget*, QFlags<Qt::WindowType>) + 87
4   libQtGui.4.dylib              	0x00000001105171cb QFrame::QFrame(QWidget*, QFlags<Qt::WindowType>) + 139
5   libqwt.6.dylib                	0x00000001182ea36d QwtTextLabel::QwtTextLabel(QWidget*) + 29
6   libqwt.6.dylib                	0x00000001182f416e QwtPlot::initPlot(QwtText const&) + 126

```

dholl, 2017-05-11:
Hmm, I'm running into the same symptoms you mentioned in https://github.com/cfriedt/gnuradio-for-mac-without-macports/issues/16#issuecomment-285972258

Did you have any tips or uncommitted patches I could start from?  Or if this is still an open issue, I don't mind poking it with a stick for a while.  :)

cfriedt, 2017-05-12:
Yep, you should be fine if you rebase from upstream master. However, I have noticed that the Qt4 Range widget is not using the specified step size. Super lame.

The other issue I ran into was in #22 and #26 - they are showstoppers for me at the moment.

dholl, 2017-05-16:
hrm...   to aid in debugging, does build.sh need plumbing to enable debug flags/switches/incantations upon all packages?  that way, maybe it could be run in a debugger to get a full backtrace, variable dump, etc...

also, by any chance, have you explored compiling the gnuradio's "next" branch with QT5 / QWT6 -- as alluded to here: https://www.bastibl.net/gnuradio-vs-qwt/

I haven't poked at any of this yet, so I have no idea how fruitful/fruitless it might be.

cfriedt, 2017-05-16:
I'd rather wait until they have a tagged release to look into building next, but haven't set debug flags for any packages.

Right now I'm working on issue #13 and it is pretty horrible.

cfriedt, 2017-07-15:
I haven't noticed anything unusual with Qt for some time, so closing this (for now).

closed 2017-07-15 by cfriedt