ktemkin/gnuradio-for-mac-without-macports #22: E300 Magically Enabled and Breaks UHD

cfriedt, 2017-05-06:
I do not think that Mac is capable of supporting the E300, which, AFAIR, is an embedded SDR on an ARM inside of an FPGA that runs Linux. Somehow, UHD detected that my Mac was capable of being an E300.. explicitly disabling E300 support for now.

Also, presumably due to some filesystem corruption and TimeMachine recovery that took place a week ago, the changes in PR #18 somehow resurfaced. It's as if those changes were lost in git, so I am reincorporating them here.

cfriedt, 2017-05-07:
```
Process:               Python [4712]
Path:                  /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python
Identifier:            Python
Version:               2.7.10 (2.7.10)
Code Type:             X86-64 (Native)
Parent Process:        Python [4536]
Responsible:           Python [4712]
User ID:               501

PlugIn Path:             /Applications/GNURadio.app/Contents/MacOS/usr/lib/libgnuradio-blocks.3.7.10.1.dylib
PlugIn Identifier:       libgnuradio-blocks.3.7.10.1.dylib
PlugIn Version:          ??? (0)

Date/Time:             2017-05-07 08:21:15.220 -0400
OS Version:            Mac OS X 10.12.3 (16D32)
Report Version:        12
Anonymous UUID:        C8698D5C-492E-BC77-25FB-DE9C5A75188D

Time Awake Since Boot: 47000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000312e35390000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [0]

VM Regions Near 0x312e35390000:
    MALLOC_LARGE (reserved 000000011aeae000-000000011b4d6000 [ 6304K] rw-/rwx SM=NUL  reserved VM address space (unallocated)
--> 
    STACK GUARD            000070000a8fe000-000070000a8ff000 [    4K] ---/rwx SM=NUL  stack guard for thread 1

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libgnuradio-blocks.3.7.10.1.dylib	0x000000010e8fdced boost::asio::detail::task_io_service::init_task() + 141
1   libgnuradio-blocks.3.7.10.1.dylib	0x000000010e8ffd68 boost::asio::io_service::service* boost::asio::detail::service_registry::create<boost::asio::datagram_socket_service<boost::asio::ip::udp> >(boost::asio::io_service&) + 120
2   libuhd.003.dylib              	0x000000011738be67 boost::asio::detail::service_registry::do_use_service(boost::asio::io_service::service::key const&, boost::asio::io_service::service* (*)(boost::asio::io_service&)) + 263
3   libuhd.003.dylib              	0x0000000117643699 udp_simple_impl::udp_simple_impl(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool, bool) + 1881
4   libuhd.003.dylib              	0x0000000117642d37 uhd::transport::udp_simple::make_broadcast(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 55
5   libuhd.003.dylib              	0x000000011739df5f usrp2_find(uhd::device_addr_t const&) + 2239
6   libuhd.003.dylib              	0x000000011739e3d9 usrp2_find(uhd::device_addr_t const&) + 3385
7   libuhd.003.dylib              	0x0000000117358dc1 boost::detail::function::function_invoker1<std::__1::vector<uhd::device_addr_t, std::__1::allocator<uhd::device_addr_t> > (*)(uhd::device_addr_t const&), std::__1::vector<uhd::device_addr_t, std::__1::allocator<uhd::device_addr_t> >, uhd::device_addr_t const&>::invoke(boost::detail::function::function_buffer&, uhd::device_addr_t const&) + 17
8   libuhd.003.dylib              	0x0000000117651618 uhd::device::find(uhd::device_addr_t const&, uhd::device::device_filter_t) + 248
9   libgnuradio-osmosdr.0.1.4.dylib	0x0000000116d261ac uhd_source_c::get_devices() + 156
10  libgnuradio-osmosdr.0.1.4.dylib	0x0000000116cd593d source_impl::source_impl(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 4285
11  libgnuradio-osmosdr.0.1.4.dylib	0x0000000116cd476f osmosdr::source::make(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 47
12  _osmosdr_swig.so              	0x0000000116c1f2b8 _wrap_source_make(_object*, _object*, _object*) + 200
13  org.python.python             	0x000000010d77010c PyEval_EvalFrameEx + 13560
14  org.python.python             	0x000000010d76ca1e PyEval_EvalCodeEx + 1617
15  org.python.python             	0x000000010d773442 0x10d6ea000 + 562242
16  org.python.python             	0x000000010d76feae PyEval_EvalFrameEx + 12954
17  org.python.python             	0x000000010d76ca1e PyEval_EvalCodeEx + 1617
18  org.python.python             	0x000000010d712531 0x10d6ea000 + 165169
19  org.python.python             	0x000000010d6f475b PyObject_Call + 99
20  org.python.python             	0x000000010d6ff40c 0x10d6ea000 + 87052
21  org.python.python             	0x000000010d6f475b PyObject_Call + 99
22  org.python.python             	0x000000010d73b283 0x10d6ea000 + 332419
23  org.python.python             	0x000000010d736ad8 0x10d6ea000 + 314072
24  org.python.python             	0x000000010d6f475b PyObject_Call + 99
25  org.python.python             	0x000000010d76f7eb PyEval_EvalFrameEx + 11223
26  org.python.python             	0x000000010d76ca1e PyEval_EvalCodeEx + 1617
27  org.python.python             	0x000000010d773442 0x10d6ea000 + 562242
28  org.python.python             	0x000000010d76feae PyEval_EvalFrameEx + 12954
29  org.python.python             	0x000000010d76ca1e PyEval_EvalCodeEx + 1617
30  org.python.python             	0x000000010d76c3c7 PyEval_EvalCode + 48
[... 600 lines cut ...]

Model: MacBookPro8,2, BootROM MBP81.0047.B32, 4 processors, Intel Core i7, 2 GHz, 16 GB, SMC 1.69f4
[... cut ...]
Network Service: Ethernet, Ethernet, en0
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: KINGSTON SV300S37A480G, 480.1 GB
Serial ATA Device: KINGSTON SV300S37A480G, 480.1 GB
USB Device: USB 2.0 Bus
USB Device: FaceTime HD Camera (Built-in)
USB Device: Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM2070 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: RTL2832U
USB Device: IR Receiver
Thunderbolt Bus: MacBook Pro, Apple Inc., 22.1

```

cfriedt, 2017-05-07:
Injected some debug messages. It *seems* like the UHD code is working.. I wonder if it's actually just GNR..

```
class udp_simple_impl : public udp_simple{
public:
    udp_simple_impl(
        const std::string &addr, const std::string &port, bool bcast, bool conn$
    ):_connected(connect){
        UHD_MSG( status ) << boost::format("Creating udp transport for %s %s") $

        //resolve the address
        asio::ip::udp::resolver resolver(_io_service);

        UHD_MSG( status ) << "Created resolver" << std::endl;

        asio::ip::udp::resolver::query query(asio::ip::udp::v4(), addr, port);

        UHD_MSG( status ) << "Created query" << std::endl;

        asio::ip::udp::resolver::iterator it = resolver.resolve(query);
        _send_endpoint = *it;
        UHD_MSG( status ) << "Resolver succeeded" << std::endl;

        //create and open the socket
        UHD_MSG( status ) << "Creating socket pointer" << std::endl;
        asio::ip::udp::socket* socketp = new asio::ip::udp::socket(_io_service)$
        UHD_MSG( status ) << "socketp: " << (void *) socketp << std::endl;
        UHD_MSG( status ) << "Created socket sptr" << std::endl;
        _socket = socket_sptr( socketp );
        _socket->open(asio::ip::udp::v4());
        UHD_MSG( status ) << "Opened socket" << std::endl;

        //allow broadcasting
        _socket->set_option(asio::socket_base::broadcast(bcast));
        UHD_MSG( status ) << "Allow broadcast on socket" << std::endl;

        //connect the socket
        if (connect) {
                _socket->connect(_send_endpoint);
                UHD_MSG( status ) << "Connected socket" << std::endl;
        }

    }
```

```
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.10.1
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy 
-- Creating udp transport for 192.168.1.255 49152
-- Created resolver
-- Created query
-- Resolver succeeded
-- Creating socket pointer
-- socketp: 0x7f8983b94ac0
-- Created socket sptr
-- Opened socket
-- Allow broadcast on socket
-- Creating udp transport for 192.168.195.255 49152
-- Created resolver
-- Created query
-- Resolver succeeded
-- Creating socket pointer
-- socketp: 0x7f89817f0ba0
-- Created socket sptr
-- Opened socket
-- Allow broadcast on socket
-- Creating udp transport for 172.16.1.255 49152
-- Created resolver
-- Created query
-- Resolver succeeded
-- Creating socket pointer
-- socketp: 0x7f8983b93fe0
-- Created socket sptr
-- Opened socket
-- Allow broadcast on socket
-- Creating udp transport for 192.168.1.255 21761
-- Created resolver
-- Created query
-- Resolver succeeded
-- Creating socket pointer
-- socketp: 0x7f8983b93aa0
-- Created socket sptr
-- Opened socket
-- Allow broadcast on socket
-- Creating udp transport for 192.168.1.255 21761
-- Created resolver
-- Created query
-- Resolver succeeded
-- Creating socket pointer
-- socketp: 0x7f8981534760
-- Created socket sptr
-- Opened socket
-- Allow broadcast on socket
-- Creating udp transport for 192.168.195.255 21761
-- Created resolver
-- Created query
-- Resolver succeeded
-- Creating socket pointer
-- socketp: 0x7f8983e8e660
-- Created socket sptr
-- Opened socket
-- Allow broadcast on socket
-- Creating udp transport for 192.168.195.255 21761
-- Created resolver
-- Created query
-- Resolver succeeded
-- Creating socket pointer
```

It looks as though the udp_simpl_impl constructor succeeds a number of times, in the last instance, it fails upon the call to `new asio::ip::udp::socket(_io_service)`. [The socket constructor](http://www.boost.org/doc/libs/1_63_0/doc/html/boost_asio/reference.html#boost_asio.reference.basic_socket.basic_socket.overload1)  is apparently never supposed to fail, but it very clearly does in this case.

cfriedt, 2017-05-10:
Ok - this is extremely weird. When I disable WiFi and do not have my ethernet plugged in, there is no segfault. Everything works as expected.

cfriedt, 2017-05-10:
This bug will remain open until the socket / service issue is resolved.

dholl, 2017-05-12:
eeek!  a crash only when enabling wifi?  hmm, and you get a crash if you plug in your wired ethernet interface, too?

cfriedt, 2017-05-12:
I think so. yes.

cfriedt, 2017-05-13:
Here's another stack trace
```
Process:               Python [5327]
Path:                  /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python
Identifier:            Python
Version:               2.7.10 (2.7.10)
Code Type:             X86-64 (Native)
Parent Process:        Python [5174]
Responsible:           Python [5327]
User ID:               501

PlugIn Path:             /Applications/GNURadio.app/Contents/MacOS/usr/lib/libgnuradio-blocks.3.7.10.1.dylib
PlugIn Identifier:       libgnuradio-blocks.3.7.10.1.dylib
PlugIn Version:          ??? (0)

Date/Time:             2017-05-12 20:20:40.161 -0400
OS Version:            Mac OS X 10.12.4 (16E195)
Report Version:        12
Anonymous UUID:        C8698D5C-492E-BC77-25FB-DE9C5A75188D

Sleep/Wake UUID:       1054C70A-B5A0-4009-AED5-7390355BFDDF

Time Awake Since Boot: 42000 seconds
Time Since Wake:       2600 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [0]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libgnuradio-blocks.3.7.10.1.dylib	0x0000000110231ced boost::asio::detail::task_io_service::init_task() + 141
1   libgnuradio-blocks.3.7.10.1.dylib	0x0000000110233d68 boost::asio::io_service::service* boost::asio::detail::service_registry::create<boost::asio::datagram_socket_service<boost::asio::ip::udp> >(boost::asio::io_service&) + 120
2   libuhd.003.dylib              	0x00000001165152c7 boost::asio::detail::service_registry::do_use_service(boost::asio::io_service::service::key const&, boost::asio::io_service::service* (*)(boost::asio::io_service&)) + 263
3   libuhd.003.dylib              	0x00000001167ccc69 udp_simple_impl::udp_simple_impl(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool, bool) + 2249
4   libuhd.003.dylib              	0x00000001167cc197 uhd::transport::udp_simple::make_broadcast(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 55
5   libuhd.003.dylib              	0x00000001165273bf usrp2_find(uhd::device_addr_t const&) + 2239
6   libuhd.003.dylib              	0x0000000116527839 usrp2_find(uhd::device_addr_t const&) + 3385
7   libuhd.003.dylib              	0x00000001164e2221 boost::detail::function::function_invoker1<std::__1::vector<uhd::device_addr_t, std::__1::allocator<uhd::device_addr_t> > (*)(uhd::device_addr_t const&), std::__1::vector<uhd::device_addr_t, std::__1::allocator<uhd::device_addr_t> >, uhd::device_addr_t const&>::invoke(boost::detail::function::function_buffer&, uhd::device_addr_t const&) + 17
8   libuhd.003.dylib              	0x00000001167db2d8 uhd::device::find(uhd::device_addr_t const&, uhd::device::device_filter_t) + 248
9   libgnuradio-osmosdr.0.1.4.dylib	0x0000000115eaf1ac uhd_source_c::get_devices() + 156
10  libgnuradio-osmosdr.0.1.4.dylib	0x0000000115e5e93d source_impl::source_impl(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 4285
11  libgnuradio-osmosdr.0.1.4.dylib	0x0000000115e5d76f osmosdr::source::make(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 47
12  _osmosdr_swig.so              	0x0000000115da82b8 _wrap_source_make(_object*, _object*, _object*) + 200
13  org.python.python             	0x000000010cbfa0ac PyEval_EvalFrameEx + 13560
14  org.python.python             	0x000000010cbf69be PyEval_EvalCodeEx + 1617
15  org.python.python             	0x000000010cbfd3e2 0x10cb74000 + 562146
16  org.python.python             	0x000000010cbf9e4e PyEval_EvalFrameEx + 12954
17  org.python.python             	0x000000010cbf69be PyEval_EvalCodeEx + 1617
18  org.python.python             	0x000000010cb9c4d1 0x10cb74000 + 165073
19  org.python.python             	0x000000010cb7e6fb PyObject_Call + 99
20  org.python.python             	0x000000010cb893ac 0x10cb74000 + 86956
21  org.python.python             	0x000000010cb7e6fb PyObject_Call + 99
22  org.python.python             	0x000000010cbc5223 0x10cb74000 + 332323
23  org.python.python             	0x000000010cbc0a78 0x10cb74000 + 313976
24  org.python.python             	0x000000010cb7e6fb PyObject_Call + 99
25  org.python.python             	0x000000010cbf978b PyEval_EvalFrameEx + 11223
26  org.python.python             	0x000000010cbf69be PyEval_EvalCodeEx + 1617
27  org.python.python             	0x000000010cbfd3e2 0x10cb74000 + 562146
28  org.python.python             	0x000000010cbf9e4e PyEval_EvalFrameEx + 12954
29  org.python.python             	0x000000010cbf69be PyEval_EvalCodeEx + 1617
30  org.python.python             	0x000000010cbf6367 PyEval_EvalCode + 48
31  org.python.python             	0x000000010cc165dd 0x10cb74000 + 665053
[... 520 lines cut ...]
===========                     =======  ======= 
TOTAL                            544.9M      908 

Model: MacBookPro8,2, BootROM MBP81.0047.B32, 4 processors, Intel Core i7, 2 GHz, 16 GB, SMC 1.69f4
[... cut ...]
Bluetooth: Version 5.0.4f18, 3 services, 17 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: KINGSTON SV300S37A480G, 480.1 GB
Serial ATA Device: KINGSTON SV300S37A480G, 480.1 GB
USB Device: USB 2.0 Bus
USB Device: FaceTime HD Camera (Built-in)
USB Device: Hub
USB Device: RTL2832U
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM2070 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: IR Receiver
Thunderbolt Bus: MacBook Pro, Apple Inc., 22.1
```

cfriedt, 2017-05-13:
Possibly also due to boost version, although (like with everything boost) tracking things is nearly impossible. In MacPorts, the boost version in use is only 1.59.0 .

I'm going to try that version and also try 1.64.0 which was released last month.

dholl, 2017-06-02:
@gilbertoverg may be having similar trouble as described in #34.  Have you had any luck?  To reproduce this crash, would I need any special hardware, or could is there anything I could run which doesn't require external parts?

gilbertoverg, 2017-06-29:
I'm getting a python crash every time I try to run a flowgraph with wifi enabled. I'm able to workaround this issue by specifying any device argument, like this for the hackrf:
```
hackrf,buffers=32
```
[image]

dholl, 2017-06-30:
@gilbertoverg Do you have a minimal .grc file you could attach that reproduces the problem, but doesn't require any special hardware?  (Is that possible?)

gilbertoverg, 2017-06-30:
@dholl  a receiver or a transmitter is required to reproduce this problem. Yesterday I was working on this flowgraph: 
[FMtransmitter.grc.zip](https://github.com/cfriedt/gnuradio-for-mac-without-macports/files/1114301/FMtransmitter.grc.zip)

michaelld, 2017-06-30:
FYI, as per https://github.com/gnuradio/gnuradio/issues/1347 , it looks like updating UHD and/or GNU Radio to the latest GIT master fixes this issue, at least for Boost 1.59.0. We don't know why yet, nor if updating just UHD or GNU Radio does the trick.

cfriedt, 2017-07-02:
@michaelld - Thanks for the tip!

closed 2017-07-02 by cfriedt