ktemkin/gnuradio-for-mac-without-macports #25: ImportError: No module named _osmosdr_swig

qunanan, 2017-05-09:
I know there isn't gr-osmosdr installed in my Mac. Do I need install it manually? Please Help. 
When I run the flowgraph it shows these...
```
Traceback (most recent call last):
  File "/Users/quxiangnan/develop/gnuradio/top_block.py", line 27, in <module>
    import osmosdr
  File "/Applications/GNURadio.app/Contents/MacOS/usr/lib/python2.7/site-packages/osmosdr/__init__.py", line 26, in <module>
    from osmosdr_swig import *
  File "/Applications/GNURadio.app/Contents/MacOS/usr/lib/python2.7/site-packages/osmosdr/osmosdr_swig.py", line 17, in <module>
    _osmosdr_swig = swig_import_helper()
  File "/Applications/GNURadio.app/Contents/MacOS/usr/lib/python2.7/site-packages/osmosdr/osmosdr_swig.py", line 16, in swig_import_helper
    return importlib.import_module('_osmosdr_swig')
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named _osmosdr_swig
```

cfriedt, 2017-05-09:
Hmm... I'm not getting that error at the moment and I'm using osmosdr for an RTL dongle.

Are you starting up GRC using the .app launcher or are you starting it from the command line.

You can start it from the command line using /Applications/GNURadio.app/Contents/MacOS/use/bin/run-grc

That's just a script that sources gr-env.sh before running gnuradio-companion, in order to set the necessary environment variables.

qunanan, 2017-05-10:
Thank you for your help.
I'm using osmosdr for HackRF.
I have tried starting GRC from the command line, it shows some WARNING messages as below
```
~ /Applications/GNURadio.app/Contents/MacOS/usr/bin/run-grc 
~ 
** (process:59907): WARNING **: Trying to register gtype 'GMountMountFlags' as enum when in fact it is of type 'GFlags'

** (process:59907): WARNING **: Trying to register gtype 'GDriveStartFlags' as enum when in fact it is of type 'GFlags'

** (process:59907): WARNING **: Trying to register gtype 'GSocketMsgFlags' as enum when in fact it is of type 'GFlags'
<<< Welcome to GNU Radio Companion 3.7.10.1 >>>
```
When I run the program, it shows the messages the same as I saw in the GRC before.

cfriedt, 2017-05-10:
Hmm... maybe I just need to upload the .dmg that I have. I thought it was the same, but maybe not.

qunanan, 2017-05-10:
OK, I see... This is caused by the environment variables are not setup correctly.
I found osmosdr in /Applications/GNURadio.app/Contents/MacOS/usr/lib/python2.7/site-packages
Which environment variable have to be set?

cfriedt, 2017-05-10:
Yes, I'm uploading a new file. You could also update the grenv.sh script manually.

Try the attachment (rename .txt to .sh and put in /Applications/GNURadio.app/Contents/MacOS/usr/bin
[grenv.txt](https://github.com/cfriedt/gnuradio-for-mac-without-macports/files/988528/grenv.txt)
)

Alternatively, you could just git clone the repo and build it yourself. It will take a while, but that way you can modify things yourself as necessary.

cfriedt, 2017-05-23:
Closing due to inactivity

closed 2017-05-23 by cfriedt