ktemkin/gnuradio-for-mac-without-macports #26: UHD SIGSEGV somewhere in rpc_client constructor

cfriedt, 2017-05-10:
```
Process:               Python [2524]
Path:                  /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python
Identifier:            Python
Version:               2.7.10 (2.7.10)
Code Type:             X86-64 (Native)
Parent Process:        Python [2331]
Responsible:           Python [2524]
User ID:               501

PlugIn Path:             /Applications/GNURadio.app/Contents/MacOS/usr/lib/libgnuradio-blocks.3.7.10.1.dylib
PlugIn Identifier:       libgnuradio-blocks.3.7.10.1.dylib
PlugIn Version:          ??? (0)

Date/Time:             2017-05-09 21:18:38.264 -0400
OS Version:            Mac OS X 10.12.4 (16E195)
Report Version:        12
Anonymous UUID:        C8698D5C-492E-BC77-25FB-DE9C5A75188D

Sleep/Wake UUID:       942FF593-FEF1-4559-B25E-8159E200A7CD

Time Awake Since Boot: 13000 seconds
Time Since Wake:       4400 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000501004000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [0]

VM Regions Near 0x501004000000:
    Dispatch continuations 000000010ce00000-000000010de00000 [ 16.0M] rw-/rwx SM=PRV  
--> 
    STACK GUARD            000070000e380000-000070000e381000 [    4K] ---/rwx SM=NUL  stack guard for thread 1

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libgnuradio-blocks.3.7.10.1.dylib	0x0000000105d1eced boost::asio::detail::task_io_service::init_task() + 141
1   libgnuradio-blocks.3.7.10.1.dylib	0x0000000105d1fc68 boost::asio::io_service::service* boost::asio::detail::service_registry::create<boost::asio::stream_socket_service<boost::asio::ip::tcp> >(boost::asio::io_service&) + 120
2   libuhd.003.dylib              	0x000000010bf6f2c7 boost::asio::detail::service_registry::do_use_service(boost::asio::io_service::service::key const&, boost::asio::io_service::service* (*)(boost::asio::io_service&)) + 263
3   libuhd.003.dylib              	0x000000010c1bccb3 uhd::usrprio_rpc::rpc_client::rpc_client(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, unsigned int, unsigned int) + 163
4   libuhd.003.dylib              	0x000000010c1c5cb5 uhd::usrprio_rpc::usrprio_rpc_client::usrprio_rpc_client(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) + 53
5   libuhd.003.dylib              	0x000000010c1c9808 uhd::niusrprio::niusrprio_session::enumerate(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::vector<uhd::usrprio_rpc::usrprio_device_info, std::__1::allocator<uhd::usrprio_rpc::usrprio_device_info> >&) + 120
6   libuhd.003.dylib              	0x000000010c07f2a7 x300_find_pcie(uhd::device_addr_t const&, bool) + 311
7   libuhd.003.dylib              	0x000000010c07d151 x300_find(uhd::device_addr_t const&) + 3073
8   libuhd.003.dylib              	0x000000010bf3c221 boost::detail::function::function_invoker1<std::__1::vector<uhd::device_addr_t, std::__1::allocator<uhd::device_addr_t> > (*)(uhd::device_addr_t const&), std::__1::vector<uhd::device_addr_t, std::__1::allocator<uhd::device_addr_t> >, uhd::device_addr_t const&>::invoke(boost::detail::function::function_buffer&, uhd::device_addr_t const&) + 17
9   libuhd.003.dylib              	0x000000010c2352d8 uhd::device::find(uhd::device_addr_t const&, uhd::device::device_filter_t) + 248
10  libgnuradio-osmosdr.0.1.4.dylib	0x000000010b9091ac uhd_source_c::get_devices() + 156
11  libgnuradio-osmosdr.0.1.4.dylib	0x000000010b8b893d source_impl::source_impl(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 4285
12  libgnuradio-osmosdr.0.1.4.dylib	0x000000010b8b776f osmosdr::source::make(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 47
13  _osmosdr_swig.so              	0x000000010b8022b8 _wrap_source_make(_object*, _object*, _object*) + 200
14  org.python.python             	0x00000001026560ac PyEval_EvalFrameEx + 13560
15  org.python.python             	0x00000001026529be PyEval_EvalCodeEx + 1617
16  org.python.python             	0x00000001026593e2 0x1025d0000 + 562146
17  org.python.python             	0x0000000102655e4e PyEval_EvalFrameEx + 12954
18  org.python.python             	0x00000001026529be PyEval_EvalCodeEx + 1617
19  org.python.python             	0x00000001025f84d1 0x1025d0000 + 165073
20  org.python.python             	0x00000001025da6fb PyObject_Call + 99
21  org.python.python             	0x00000001025e53ac 0x1025d0000 + 86956
22  org.python.python             	0x00000001025da6fb PyObject_Call + 99
23  org.python.python             	0x0000000102621223 0x1025d0000 + 332323
24  org.python.python             	0x000000010261ca78 0x1025d0000 + 313976
25  org.python.python             	0x00000001025da6fb PyObject_Call + 99
26  org.python.python             	0x000000010265578b PyEval_EvalFrameEx + 11223
27  org.python.python             	0x00000001026529be PyEval_EvalCodeEx + 1617
[... 525 lines cut ...]
TOTAL                            560.9M      910 

Model: MacBookPro8,2, BootROM MBP81.0047.B32, 4 processors, Intel Core i7, 2 GHz, 16 GB, SMC 1.69f4
Graphics: Intel HD Graphics 3000, Intel HD Graphics 3000, Built-In
Graphics: AMD Radeon HD 6490M, AMD Radeon HD 6490M, PCIe, 256 MB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1333 MHz, 0x04CD, 0x46412D3133333343392D3847535100000020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1333 MHz, 0x04CD, 0x46412D3133333343392D3847535100000020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xD6), Broadcom BCM43xx 1.0 (5.106.98.100.24)
Bluetooth: Version 5.0.4f18, 3 services, 17 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: KINGSTON SV300S37A480G, 480.1 GB
Serial ATA Device: KINGSTON SV300S37A480G, 480.1 GB
USB Device: USB 2.0 Bus
USB Device: FaceTime HD Camera (Built-in)
USB Device: Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM2070 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: RTL2832U
USB Device: IR Receiver
Thunderbolt Bus: MacBook Pro, Apple Inc., 22.1

```

michaelld, 2017-06-30:
FYI, as per https://github.com/gnuradio/gnuradio/issues/1347 , it looks like updating UHD and/or GNU Radio to the latest GIT master fixes this issue, at least for Boost 1.59.0. We don't know why yet, nor if updating just UHD or GNU Radio does the trick.

cfriedt, 2017-07-02:
@michaelld - Thanks for the tip!

closed 2017-07-02 by cfriedt