ktemkin/gnuradio-for-mac-without-macports #28: checksums of downloaded files?

dholl, 2017-05-11:
I'm sometimes too paranoid for my own good.  But would you be interested if I conjured a way for build.sh to verify SHA-256 checksums of each downloaded file?  I figure that since the download URL's are hardcoded into build.sh, the expected checksum could be hardcoded as well.  For example:

```
P=qt-everywhere-opensource-src-4.8.7
URL=https://www.mirrorservice.org/sites/download.qt-project.org/official_releases/qt/4.8/4.8.7/qt-everywhere-opensource-src-4.8.7.tar.gz
SHA256="e2882295097e47fe089f8ac741a95fef47e0a73a3f3cdf21b56990638f626ea0"
T=${P}
BRANCH=""
```

and I'd modify fetch() to verify the checksum before returning.

cfriedt, 2017-05-12:
That would be great. I would suggest making another function, e.g. sha256(), with the first argument being a filename.

dholl, 2017-05-13:
At the moment, I'm trying an idea using this syntax for regular files:
```
CKSUM=sha256:e2882295097e47fe089f8ac741a95fef47e0a73a3f3cdf21b56990638f626ea0
```

and git git repos, something like:
```
CKSUM=git:59daaff0d9d04373d3a6b14ea7b46e080bad7a1e
```

In the case of git repos, the hash is just the full sha hash from "git log", and the only verification I'm performing is that the hash refers to a commit that is somewhere in the history of the selected branch, so that newer commits do not invalidate the hash.  Does that sound all right?  Or if we'd want this to be more stringent, then we could always "checkout" the exact hash.

Here's a snapshot of the verification code at the moment:  (I'm still digging up checksums for the various packages, and plan to re-run the whole build once I have them all in.)
```sh
function verify_sha256() {
  #local FILENAME="${1}"
  #local CKSUM="${2}"
  test "$(shasum -a 256 -- "${1}" | cut -d' ' -f1)" == "${2}"
}

function verify_git() {
  #local FILENAME="${1}"
  #local CKSUM="${2}"
  # Verify the hash refers to a commit.
  #    http://stackoverflow.com/questions/18515488/how-to-check-if-the-commit-exists-in-a-git-repository-by-its-sha-1
  # Then verify the hash is in the current branch.  (The branch may have newer commits.)
  #    http://stackoverflow.com/questions/4127967/validate-if-commit-exists
  (
  cd "${1}" \
    && test "$(git cat-file -t "${2}" 2>/dev/null)" = commit \
    && test "$( git rev-list HEAD.."${2}" | wc -l )" -eq 0
  )
  # directory is restored at )
}

function verify_checksum() {
  #local FILENAME="${1}"
  #local CKSUM="${2}"
  test -e "${1}" || E "Missing: '${1}'"
  if [ -z "${2}" ]; then
    # Nag someone to get a checksum for this thing.
    I "No checksum: '${1}'"
    return 0
  elif [ "${2%%:*}" = "sha256" ]; then
    # Strip off the leading sha256: and invoke the correct function:
    verify_sha256 "${1}" "${2#*:}" \
      && D "File '${1}' matches '${2}'" \
      || E "File '${1}' does not match '${2}'"
  elif [ "${2%%:*}" = "git" ]; then
    # Strip off the leading git: and invoke the correct function:
    verify_git "${1}" "${2#*:}" \
      && D "Repository '${1}' matches '${2}'" \
      || E "Repository '${1}' does not match '${2}'"
  else
    E "Unrecognized checksum format: ${2}"
  fi
}
```

```verify_checksum``` gets executed invoked from ```fetch```, which I've tweaked to accept a fifth input argument:
```sh
function fetch() {
  local P=${1}
  local URL=${2}
  local T=${3}
  local BRANCH=${4}
  local CKSUM="${5}"
```

So at the moment, I'm just running through with the rest of the plumbing.

cfriedt, 2017-05-13:
Well done!

cfriedt, 2017-05-27:
fixed in #31

closed 2017-05-27 by cfriedt