ktemkin/gnuradio-for-mac-without-macports #33: verify_checksum for git seems broken

cfriedt, 2017-05-31:
```
I: fetching libzmq from git://github.com/zeromq/libzmq.git
D: downloading to /Users/jdoe/workspace/gnuradio-for-mac-without-macports/tmp/libzmq
D: already downloaded libzmq
E: Repository '/Users/jdoe/workspace/gnuradio-for-mac-without-macports/tmp/libzmq' does not match 'd17581929cceceda02b4eb8abb054f996865c7a6'
```

cfriedt, 2017-05-31:
@dholl - any idea?

dholl, 2017-05-31:
Hmm, I'm not seeing the error on my end.  Too bad I used the same error message
```sh
E "Repository '${1}' does not match '${2}'"
```
twice, for both lines 115 and 119.

Could you try running these from your command line:
```sh
# This should print "commit"
git -C /Users/jdoe/workspace/gnuradio-for-mac-without-macports/tmp/libzmq \
    cat-file -t d17581929cceceda02b4eb8abb054f996865c7a6
# This should show 0:
printf 'cat-file status: %s\n' "$?"
# This should print nothing:
git -C /Users/jdoe/workspace/gnuradio-for-mac-without-macports/tmp/libzmq \
    rev-list HEAD..d17581929cceceda02b4eb8abb054f996865c7a6
# This should also show 0:
printf 'rev-list status: %s\n' "$?"
# Are you on branch "master"?
git -C /Users/jdoe/workspace/gnuradio-for-mac-without-macports/tmp/libzmq \
    branch
```

Do you happen to have a different branch checked out?  (verify ```git branch``` reports master?)

Also, this is unrelated, but upon checking ```git log | head -n1```, I found that libzmq does have more recent commits than d17581929cceceda02b4eb8abb054f996865c7a6.  Currently, my updated clone shows bcc30f2ab0b91cffc94a667588f0ef7153993942 as the most recent commit, but the checks still pass for me without errors, as intended.  But this is one of the cases when that Notice regarding newer commits would have lit up to alert us that whatever we're building may differ due to upstream activity in libzmq...  ;)

cfriedt, 2017-06-04:
@dholl build.sh may very well have created a "local" branch prior to applying git patches. That is expected behaviour.

dholl, 2017-06-05:
Um, if a branch was created (even an unnamed one), then it will still carry over the history of previous commits from the current branch that was checked out, and thus commit d17581929cceceda02b4eb8abb054f996865c7a6 should still be there.  Right?

cfriedt, 2017-06-05:
I'm not actually. I wonder if there actually was some corruption somewhere along the lines... or maybe... I was hacked!!!! O.o

dholl, 2017-06-06:
> I'm not actually.

Whew.  :)

> I wonder if there actually was some corruption somewhere along the lines... or maybe... I was hacked!!!! O.o

All the more reason for the checksums.  ;)

cfriedt, 2017-06-24:
Maybe I'll just close it.

closed 2017-06-24 by cfriedt