ktemkin/gnuradio-for-mac-without-macports #34: osmocom_fft and osmocom_siggen

gilbertoverg, 2017-05-31:
Hi, I can see that osmocom_fft and osmocom_siggen are present inside /GNURadio.app/Contents/MacOS/usr/
How can I run both?
Thank you

dholl, 2017-06-01:
To access the other binaries under ```usr/bin```, there are a few environment variables to set first -- specified in ```grenv.sh```.  From a shell (sh, ash, bash, dash, ksh, zsh, ..., anything except for csh/tcsh), try this first:
```sh
. /Applications/GNURadio.app/Contents/MacOS/usr/bin/grenv.sh
```
The ```.``` on the first line tells your shell to source the contents of ```grenv.sh```.  You only need to source this once.  If you want, you could put this ```. /.../usr/bin/grenv.sh``` command in your shell's startup scripts, such as ```~/.zshenv``` if you're using zsh.  See the man page for your shell for details.  (Note: bash is a little complicated.  Definitely consult ```man bash``` if you're using bash; check out the section titled ```INVOCATION```.)

Now you should be able to run other programs in ```.../usr/bin```, such as
```sh
osmocom_fft --help
```

Does that work for you?

Disclaimer:  From my former sys-admin days, I've learned to appreciate zsh, especially for its simplified startup script sequence --- as highlighted in ```man zsh``` under the ```STARTUP/SHUTDOWN FILES``` section.

If you do want to modify your shell's startup files, try something like:
```sh
if test -e /Applications/GNURadio.app/Contents/MacOS/usr/bin/grenv.sh ; then
    . /Applications/GNURadio.app/Contents/MacOS/usr/bin/grenv.sh
fi
```
so that if you ever move the GNURadio.app bundle, then your shell won't complain too much.

gilbertoverg, 2017-06-02:
I've looked at run-grc and I've created run-fft:
```
#!/bin/sh
source /Applications/GNURadio.app/Contents/MacOS/usr/bin/grenv.sh
osmocom_fft ${@} &
exit 0
```
and run-siggen:
```
#!/bin/sh
source /Applications/GNURadio.app/Contents/MacOS/usr/bin/grenv.sh
osmocom_siggen ${@} &
exit 0
```
I'm getting the option list running ./run-fft -help and ./run-siggen -help
But when running ./run-fft and ./run-siggen with wifi enabled I'm getting a crash.
With wifi disabled, running ./run-fft does not work and shows this on the terminal:
```
bash-3.2$ ./run-fft
bash-3.2$ Mac OS; Clang version 8.1.0 (clang-802.0.42); Boost_106300; UHD_003.010.001.local-release_003_010_001_001-0-gc705922a

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.10.1
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy 
Using HackRF One with firmware 2017.02.1 
Traceback (most recent call last):
  File "/Applications/GNURadio.app/Contents/MacOS/usr/bin/osmocom_fft", line 841, in <module>
    main ()
  File "/Applications/GNURadio.app/Contents/MacOS/usr/bin/osmocom_fft", line 837, in main
    app = stdgui2.stdapp(app_top_block, "osmocom Spectrum Browser", nstatus=1)
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/gnuradio/wxgui/stdgui2.py", line 46, in __init__
    wx.App.__init__ (self, redirect=False)
  File "/Applications/GNURadio.app/Contents/MacOS/usr/lib/python2.7/site-packages/wx-3.0-gtk2/wx/_core.py", line 8628, in __init__
    self._BootstrapApp()
  File "/Applications/GNURadio.app/Contents/MacOS/usr/lib/python2.7/site-packages/wx-3.0-gtk2/wx/_core.py", line 8196, in _BootstrapApp
    return _core_.PyApp__BootstrapApp(*args, **kwargs)
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/gnuradio/wxgui/stdgui2.py", line 49, in OnInit
    frame = stdframe (self.top_block_maker, self.title, self._nstatus)
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/gnuradio/wxgui/stdgui2.py", line 76, in __init__
    self.panel = stdpanel (self, self, top_block_maker)
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/gnuradio/wxgui/stdgui2.py", line 98, in __init__
    self.top_block = top_block_maker (frame, self, vbox, sys.argv)
  File "/Applications/GNURadio.app/Contents/MacOS/usr/bin/osmocom_fft", line 246, in __init__
    fft_rate=options.fft_rate)
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/gnuradio/wxgui/fftsink_nongl.py", line 198, in __init__
    self.win = fft_window(self, parent, size=size)
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/gnuradio/wxgui/fftsink_nongl.py", line 331, in __init__
    self.control_panel = control_panel(self)
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/gnuradio/wxgui/fftsink_nongl.py", line 256, in __init__
    wx.Panel.__init__(self, parent, -1, style=wx.SIMPLE_BORDER)
  File "/Applications/GNURadio.app/Contents/MacOS/usr/lib/python2.7/site-packages/wx-3.0-gtk2/wx/_windows.py", line 68, in __init__
    _windows_.Panel_swiginit(self,_windows_.new_Panel(*args, **kwargs))
wx._core.PyAssertionError: C++ assertion "m_window" failed at ./src/gtk/dcclient.cpp(2043) in DoGetSize(): GetSize() doesn't work without window
```
With wifi disabled ./run-siggen works:
[image]

dholl, 2017-06-02:
I don't have any ideas for the crash when wifi is enabled.  @cfriedt any ideas?

---

Regarding the small scripts, I have some small fixes for ```run-grc``` to address ```sh``` scripting practices as well as #5 , but I'm waiting to see what @cfriedt has up his sleeve with regard to #13 before I submit my own ```run-grc``` fixes.

Your script has the right idea, but for those command line tools, try this template:  (This won't fix your crash.)
```sh
#!/bin/sh
. /Applications/GNURadio.app/Contents/MacOS/usr/bin/grenv.sh
exec osmocom_fft "${@}"
```

```sh
#!/bin/sh
. /Applications/GNURadio.app/Contents/MacOS/usr/bin/grenv.sh
exec osmocom_siggen "${@}"
```

Summary:  (Feel free to ignore or use only pieces of these tips.  Use " around ${@} to expand correctly for _all_ input arguments, such as spaces or whatever.  Use ```exec``` to replace the ```sh``` process with your binary.  Don't need ```&``` within the script, but if you'd like to run the whole thing in background, then use ```./run-fft &```.  And use ```.``` instead of ```source```, since . is in the POSIX sh spec, but source is not --- or just change the first line to ```#!/bin/bash``` to declare that you'd prefer bash syntax in the first place.  ;)

If you'd like a single script to let you run any of the commands in /Applications/GNURadio.app/Contents/MacOS/usr/bin, then try saving this file to ```$HOME/gr``` with ```chmod +x "$HOME/gr"```

```sh
#!/bin/sh
prog="$1"
# shift will drop $1 and shift over the rest of the args so $2 becomes $1, $3 becomes $2, etc...
shift
. /Applications/GNURadio.app/Contents/MacOS/usr/bin/grenv.sh
exec "${prog}" "${@}"
```

Then you should be able to run ```osmocom_fft``` or any of the utilities without needing separate scripts:
```sh
~/gr osmocom_fft --help
~/gr osmocom_siggen --help
```

The nifty part of restructuring your scripts like this (using exec, no &, and " around $@), is that now if you wanted to automate setting up your project, you can write a master script to launch your subprocesses, keeping track of their child PID's, and killing them when you're done.

```sh
#!/bin/sh

# Launch osmocom_siggen in background and save its pid:
~/gr osmocom_siggen --some --args &
pid_siggen=$!

# Launch osmocom_fft in background and save its pid:
~/gr osmocom_fft --some --other --args &
pid_fft=$!

# do other stuff, collect your data, etc...

# now shut down:
kill $pid_siggen $pid_fft
```

This job control wouldn't be possible if the ```&``` was present in subscripts like ```gr``` or ```run-fft```, because the child PID of the subprocess would've already been discarded in the sub shell.

Sorry I won't be of much help to resolve the crash when wireless is enabled, but I hope the scripting pointers will help build empires.

gilbertoverg, 2017-06-02:
@dholl thank you so much for your help

now when I try to run osmocom_siggen sometimes it works (even with wifi enabled) sometimes it crashes (even with wifi disabled), this is the crash window:
```
Process:               Python [36237]
Path:                  /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python
Identifier:            Python
Version:               2.7.10 (2.7.10)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Python [36237]
User ID:               501

PlugIn Path:             /Applications/GNURadio.app/Contents/MacOS/usr/lib/libgnuradio-blocks.3.7.10.1.dylib
PlugIn Identifier:       libgnuradio-blocks.3.7.10.1.dylib
PlugIn Version:          ??? (0)

Date/Time:             2017-06-02 19:55:03.078 +0200
OS Version:            Mac OS X 10.12.5 (16F73)
Report Version:        12
Anonymous UUID:        89217373-AFE7-7417-702C-977184FF22A8

Sleep/Wake UUID:       98686CC2-CB85-4B84-91B1-0200968BDC32

Time Awake Since Boot: 450000 seconds
Time Since Wake:       680 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000010000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [0]

VM Regions Near 0x10000:
--> 
    __TEXT                 0000000101ceb000-0000000101ced000 [    8K] r-x/rwx SM=COW  /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libgnuradio-blocks.3.7.10.1.dylib	0x0000000104880ced boost::asio::detail::task_io_service::init_task() + 141
1   libgnuradio-blocks.3.7.10.1.dylib	0x0000000104882d68 boost::asio::io_service::service* boost::asio::detail::service_registry::create<boost::asio::datagram_socket_service<boost::asio::ip::udp> >(boost::asio::io_service&) + 120
2   libuhd.003.dylib              	0x00000001082b6e67 boost::asio::detail::service_registry::do_use_service(boost::asio::io_service::service::key const&, boost::asio::io_service::service* (*)(boost::asio::io_service&)) + 263
3   libuhd.003.dylib              	0x000000010856e699 udp_simple_impl::udp_simple_impl(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool, bool) + 1881
4   libuhd.003.dylib              	0x000000010856dd37 uhd::transport::udp_simple::make_broadcast(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 55
5   libuhd.003.dylib              	0x00000001082c8f5f usrp2_find(uhd::device_addr_t const&) + 2239
6   libuhd.003.dylib              	0x00000001082c93d9 usrp2_find(uhd::device_addr_t const&) + 3385
7   libuhd.003.dylib              	0x0000000108283dc1 boost::detail::function::function_invoker1<std::__1::vector<uhd::device_addr_t, std::__1::allocator<uhd::device_addr_t> > (*)(uhd::device_addr_t const&), std::__1::vector<uhd::device_addr_t, std::__1::allocator<uhd::device_addr_t> >, uhd::device_addr_t const&>::invoke(boost::detail::function::function_buffer&, uhd::device_addr_t const&) + 17
8   libuhd.003.dylib              	0x000000010857c618 uhd::device::find(uhd::device_addr_t const&, uhd::device::device_filter_t) + 248
9   libgnuradio-osmosdr.0.1.4.dylib	0x0000000107c4adec uhd_sink_c::get_devices() + 156
10  libgnuradio-osmosdr.0.1.4.dylib	0x0000000107c14e8b sink_impl::sink_impl(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 1979
11  libgnuradio-osmosdr.0.1.4.dylib	0x0000000107c145bf osmosdr::sink::make(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 47
12  _osmosdr_swig.so              	0x0000000107b4f0d8 _wrap_sink_make(_object*, _object*, _object*) + 200
13  org.python.python             	0x0000000101d770ac PyEval_EvalFrameEx + 13560
14  org.python.python             	0x0000000101d739be PyEval_EvalCodeEx + 1617
15  org.python.python             	0x0000000101d7a3e2 0x101cf1000 + 562146
16  org.python.python             	0x0000000101d76e4e PyEval_EvalFrameEx + 12954
17  org.python.python             	0x0000000101d7a475 0x101cf1000 + 562293
18  org.python.python             	0x0000000101d76e4e PyEval_EvalFrameEx + 12954
19  org.python.python             	0x0000000101d739be PyEval_EvalCodeEx + 1617
20  org.python.python             	0x0000000101d194d1 0x101cf1000 + 165073
21  org.python.python             	0x0000000101cfb6fb PyObject_Call + 99
22  org.python.python             	0x0000000101d063ac 0x101cf1000 + 86956
23  org.python.python             	0x0000000101cfb6fb PyObject_Call + 99
24  org.python.python             	0x0000000101d42223 0x101cf1000 + 332323
25  org.python.python             	0x0000000101d3da78 0x101cf1000 + 313976
[... 534 lines cut ...]
VM_ALLOCATE                       39.5M       45 
__DATA                            25.5M      373 
__IMAGE                            528K        2 
__LINKEDIT                       142.8M      138 
__TEXT                           216.6M      339 
[... cut ...]
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x802C, 0x31364B544631473634485A2D314736453120
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x152), Broadcom BCM43xx 1.0 (7.21.171.126.1a2)
Bluetooth: Version 5.0.4f18, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0512G, 500,28 GB
USB Device: USB 3.0 Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Bluetooth USB Host Controller
USB Device: HackRF One
Thunderbolt Bus: MacBook Pro, Apple Inc., 27.1
```

dholl, 2017-06-02:
Eeck, this looks very similar to #22 .  @cfriedt , Any luck with your crash diagnostics?  Do you think your troubles and @gilbertoverg's are related?

(I develop my own hardware, and don't have other vendors' hardware to poke at the moment...  Do I need any special hardware to try reproducing this crash on my own?  What's a minimal test I could try?)

gilbertoverg, 2017-06-02:
I'm using an hackrf to test osmocom_siggen and osmocom_fft, I can also test osmocom_fft with an airspy. I'm getting the crash 1 second after running the script

gilbertoverg, 2017-06-03:
This is interesting: if I remove the two check marks from the xquartz security settings, osmocom_siggen crashes far more often than having only the "authenticate connections" enabled

gilbertoverg, 2017-06-03:
This is more interesting: I'm running wireshark listening the loopback interface, when osmocom_siggen starts correctly this is what I'm getting:
[image]
or:
[image]
When it crashes:
[image]
or:
[image]
or:
[image]

gilbertoverg, 2017-06-03:
I've temporary disabled the loopback interface with ```sudo ifconfig lo0 down```, disabled the wifi and osmocom_siggen works every time with this trick: changing the position of the window before closing it

gilbertoverg, 2017-06-03:
@dholl you can reproduce this crash without any special hardware, simply try run osmocom_siggen with no arguments, if it works it prints:
```
Mac OS; Clang version 8.1.0 (clang-802.0.42); Boost_106300; UHD_003.010.001.local-release_003_010_001_001-0-gc705922a

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.10.1
built-in sink types: uhd hackrf bladerf 

FATAL: No supported devices found to pick from.

Trying to fill up 1 missing channel(s) with null sink(s).
This is being done to prevent the application from crashing
due to gnuradio bug #528.

Sink has no sample rates (wrong device arguments?).
```
if it crashes:
```
Mac OS; Clang version 8.1.0 (clang-802.0.42); Boost_106300; UHD_003.010.001.local-release_003_010_001_001-0-gc705922a

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.10.1
built-in sink types: uhd hackrf bladerf 
Segmentation fault: 11
```

cfriedt, 2017-06-04:
I've seen this too - it's crashing somewhere either in UHD or boost when it's searching for network interfaces, if i remember correctly. Technically, I believe this duplicates either #22 or #26 .

gilbertoverg, 2017-06-04:
@cfriedt I'm still not able to run osmocom_fft, this is what I'm getting:
```
Mac OS; Clang version 8.1.0 (clang-802.0.42); Boost_106300; UHD_003.010.001.local-release_003_010_001_001-0-gc705922a

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.10.1
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy 
Using HackRF One with firmware 2017.02.1 
Traceback (most recent call last):
  File "/Applications/GNURadio.app/Contents/MacOS/usr/bin/osmocom_fft", line 841, in <module>
    main ()
  File "/Applications/GNURadio.app/Contents/MacOS/usr/bin/osmocom_fft", line 837, in main
    app = stdgui2.stdapp(app_top_block, "osmocom Spectrum Browser", nstatus=1)
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/gnuradio/wxgui/stdgui2.py", line 46, in __init__
    wx.App.__init__ (self, redirect=False)
  File "/Applications/GNURadio.app/Contents/MacOS/usr/lib/python2.7/site-packages/wx-3.0-gtk2/wx/_core.py", line 8628, in __init__
    self._BootstrapApp()
  File "/Applications/GNURadio.app/Contents/MacOS/usr/lib/python2.7/site-packages/wx-3.0-gtk2/wx/_core.py", line 8196, in _BootstrapApp
    return _core_.PyApp__BootstrapApp(*args, **kwargs)
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/gnuradio/wxgui/stdgui2.py", line 49, in OnInit
    frame = stdframe (self.top_block_maker, self.title, self._nstatus)
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/gnuradio/wxgui/stdgui2.py", line 76, in __init__
    self.panel = stdpanel (self, self, top_block_maker)
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/gnuradio/wxgui/stdgui2.py", line 98, in __init__
    self.top_block = top_block_maker (frame, self, vbox, sys.argv)
  File "/Applications/GNURadio.app/Contents/MacOS/usr/bin/osmocom_fft", line 246, in __init__
    fft_rate=options.fft_rate)
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/gnuradio/wxgui/fftsink_nongl.py", line 198, in __init__
    self.win = fft_window(self, parent, size=size)
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/gnuradio/wxgui/fftsink_nongl.py", line 331, in __init__
    self.control_panel = control_panel(self)
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/gnuradio/wxgui/fftsink_nongl.py", line 256, in __init__
    wx.Panel.__init__(self, parent, -1, style=wx.SIMPLE_BORDER)
  File "/Applications/GNURadio.app/Contents/MacOS/usr/lib/python2.7/site-packages/wx-3.0-gtk2/wx/_windows.py", line 68, in __init__
    _windows_.Panel_swiginit(self,_windows_.new_Panel(*args, **kwargs))
wx._core.PyAssertionError: C++ assertion "m_window" failed at ./src/gtk/dcclient.cpp(2043) in DoGetSize(): GetSize() doesn't work without window
```

cfriedt, 2017-06-04:
@gilbertoverg pull requests are welcome.

gilbertoverg, 2017-06-05:
@cfriedt after a long day I found a fix for osmocom_fft (I'm not a python programmer):
In /GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/gnuradio/wxgui/plot.py comment out line 497
```
self.OnSize(None) # sets the initial size based on client size
```

cfriedt, 2017-06-05:
Nice! I feel your pain - I'm also not too happy with Python, but it seems to be what all the hip kids are using these days ;-)

cfriedt, 2017-07-15:
Closing due to inactivity, and because the stack trace referenced was fixed in PR #48

closed 2017-07-15 by cfriedt