ktemkin/gnuradio-for-mac-without-macports #43: Python Crash when use libgnuradio-blocks.3.7.10.1.dylib PlugIn

cn0xroot, 2017-06-09:
Hi when i use it in osx 10.12.5 ,python crash.

```
Process:               Python [62018]
Path:                  /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python
Identifier:            Python
Version:               2.7.10 (2.7.10)
Code Type:             X86-64 (Native)
Parent Process:        Python [62003]
Responsible:           Python [62018]
User ID:               501

PlugIn Path:             /Applications/GNURadio.app/Contents/MacOS/usr/lib/libgnuradio-blocks.3.7.10.1.dylib
PlugIn Identifier:       libgnuradio-blocks.3.7.10.1.dylib
PlugIn Version:          ??? (0)

Date/Time:             2017-06-09 15:06:12.552 +0800
OS Version:            Mac OS X 10.12.5 (16F73)
Report Version:        12
```

what something error？

cfriedt, 2017-06-24:
Can you supply a bit more of the stack trace?

cn0xroot, 2017-06-26:
```
Process:               Python [2598]
Path:                  /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python
Identifier:            Python
Version:               2.7.10 (2.7.10)
Code Type:             X86-64 (Native)
Parent Process:        Python [2371]
Responsible:           Python [2598]
User ID:               501

PlugIn Path:             /Applications/GNURadio.app/Contents/MacOS/usr/lib/libgnuradio-blocks.3.7.10.1.dylib
PlugIn Identifier:       libgnuradio-blocks.3.7.10.1.dylib
PlugIn Version:          ??? (0)

Date/Time:             2017-06-26 18:15:40.025 +0800
OS Version:            Mac OS X 10.12.5 (16F73)
Report Version:        12
Anonymous UUID:        902D2FEB-357F-7C2F-6BC0-D460C494C2FC

Time Awake Since Boot: 800 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [0]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libgnuradio-blocks.3.7.10.1.dylib	0x000000010677cced boost::asio::detail::task_io_service::init_task() + 141
1   libgnuradio-blocks.3.7.10.1.dylib	0x000000010677ed68 boost::asio::io_service::service* boost::asio::detail::service_registry::create<boost::asio::datagram_socket_service<boost::asio::ip::udp> >(boost::asio::io_service&) + 120
2   libuhd.003.dylib              	0x000000010c92be67 boost::asio::detail::service_registry::do_use_service(boost::asio::io_service::service::key const&, boost::asio::io_service::service* (*)(boost::asio::io_service&)) + 263
3   libuhd.003.dylib              	0x000000010cbe3699 udp_simple_impl::udp_simple_impl(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool, bool) + 1881
4   libuhd.003.dylib              	0x000000010cbe2d37 uhd::transport::udp_simple::make_broadcast(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 55
5   libuhd.003.dylib              	0x000000010c9c7373 uhd::usrp::e300::e300_find(uhd::device_addr_t const&) + 4531
6   libuhd.003.dylib              	0x000000010c9c75e2 uhd::usrp::e300::e300_find(uhd::device_addr_t const&) + 5154
7   libuhd.003.dylib              	0x000000010c9c724a uhd::usrp::e300::e300_find(uhd::device_addr_t const&) + 4234
8   libuhd.003.dylib              	0x000000010c8f8dc1 boost::detail::function::function_invoker1<std::__1::vector<uhd::device_addr_t, std::__1::allocator<uhd::device_addr_t> > (*)(uhd::device_addr_t const&), std::__1::vector<uhd::device_addr_t, std::__1::allocator<uhd::device_addr_t> >, uhd::device_addr_t const&>::invoke(boost::detail::function::function_buffer&, uhd::device_addr_t const&) + 17
9   libuhd.003.dylib              	0x000000010cbf1618 uhd::device::find(uhd::device_addr_t const&, uhd::device::device_filter_t) + 248
10  libgnuradio-osmosdr.0.1.4.dylib	0x000000010c2c61ac uhd_source_c::get_devices() + 156
11  libgnuradio-osmosdr.0.1.4.dylib	0x000000010c27593d source_impl::source_impl(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 4285
12  libgnuradio-osmosdr.0.1.4.dylib	0x000000010c27476f osmosdr::source::make(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 47
13  _osmosdr_swig.so              	0x000000010c1bf2b8 _wrap_source_make(_object*, _object*, _object*) + 200
14  org.python.python             	0x000000010307c0ac PyEval_EvalFrameEx + 13560
15  org.python.python             	0x00000001030789be PyEval_EvalCodeEx + 1617
16  org.python.python             	0x000000010307f3e2 0x102ff6000 + 562146
17  org.python.python             	0x000000010307be4e PyEval_EvalFrameEx + 12954
18  org.python.python             	0x00000001030789be PyEval_EvalCodeEx + 1617
19  org.python.python             	0x000000010301e4d1 0x102ff6000 + 165073
20  org.python.python             	0x00000001030006fb PyObject_Call + 99
21  org.python.python             	0x000000010300b3ac 0x102ff6000 + 86956
22  org.python.python             	0x00000001030006fb PyObject_Call + 99
23  org.python.python             	0x0000000103047223 0x102ff6000 + 332323
24  org.python.python             	0x0000000103042a78 0x102ff6000 + 313976
25  org.python.python             	0x00000001030006fb PyObject_Call + 99
26  org.python.python             	0x000000010307b78b PyEval_EvalFrameEx + 11223
27  org.python.python             	0x00000001030789be PyEval_EvalCodeEx + 1617
28  org.python.python             	0x000000010307f3e2 0x102ff6000 + 562146
29  org.python.python             	0x000000010307be4e PyEval_EvalFrameEx + 12954
30  org.python.python             	0x00000001030789be PyEval_EvalCodeEx + 1617
31  org.python.python             	0x0000000103078367 PyEval_EvalCode + 48
32  org.python.python             	0x00000001030985dd 0x102ff6000 + 665053
33  org.python.python             	0x0000000103098680 PyRun_FileExFlags + 133
34  org.python.python             	0x00000001030981d1 PyRun_SimpleFileExFlags + 702
35  org.python.python             	0x00000001030a9b6a Py_Main + 3094
[... cut ...]
    0x7fffd1b61000 -     0x7fffd1b8aff7  libxpc.dylib (972.60.2) <1C9AF716-69DF-359F-85E9-7DFDE362F9A2> /usr/lib/system/libxpc.dylib

External Modification Summary:
  Calls made by other processes targeting this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by all processes on this machine:
    task_for_pid: 623
    thread_create: 0
    thread_set_state: 0

```

cfriedt, 2017-06-29:
This is a duplicate of Issue #22 and possibly #26

closed 2017-06-29 by cfriedt