ktemkin/gnuradio-for-mac-without-macports #44: Every menu and text from right to left

mehdideveloper, 2017-06-09:
Hi
I just installed this on OS X 10.12.5 and after the first run, here's what I got:

[image]

Any ideas what could've went wrong?
Thanks

dholl, 2017-06-11:
Cool!  I'm also running OS X 10.12.5, but I don't get the right-to-left menu.

Could you open a Terminal (Applications -> Utilities -> Terminal) and run ```locale```, please?

For example, my locale output looks like:
```
zero(ttys006):~> locale
LANG="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_CTYPE="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_ALL=
zero(ttys006):~>
```

dholl, 2017-06-11:
Also, this section in Apple's docs may be relevant:
"Testing Right-to-Left Locale Formats"
https://developer.apple.com/library/content/documentation/MacOSX/Conceptual/BPInternational/SupportingRight-To-LeftLanguages/SupportingRight-To-LeftLanguages.html#//apple_ref/doc/uid/10000171i-CH17-SW12

Under "System Preferences" -> "Language & Region", what Region do you have set, and then clicking on the "Advanced..." button, what is your "Format language" set to?

mehdideveloper, 2017-06-11:
Thanks for the reply.
Here's the output of locale:
`LANG="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_CTYPE="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_ALL=`

My 'Region' is 'United States' and there's no "Format Language" in the "Advanced..."  section.

mehdideveloper, 2017-06-11:
When I execute the 'locale' in XQuartz's Terminal, here's what I get:

` LANG=`
`LC_COLLATE="C"`
`LC_CTYPE="C"`
`LC_MESSAGES="C"`
`LC_MONETARY="C"`
`LC_NUMERIC="C"`
`LC_TIME="C"`
`LC_ALL=`

mehdideveloper, 2017-06-11:
I managed to solve this:
Created an environment variable called "LANG" with the value set to "en_US"
Then ran this command: "open /Applications/GNURadio.app

dholl, 2017-06-13:
Hmm, very odd that your initial locale looked identical to mine with ```en_US.UTF-8``` everywhere.
So, using ```LANG=en_US``` fixes it?  Could you try ```LANG=en_US.UTF-8```?

Also, how did you set the environment variable?  One way to set it permanently is to edit ```GNURadio.app/Contents/MacOS/usr/bin/grenv.sh``` (with a simple text editor, such as emacs, or vi...), and add
```sh
export LANG=en_US.UTF-8
```
perhaps at the end of grenv.sh, just after ```export PATH```.

I wish I could reproduce this though.  I did try running locale on my system under XQuartz, and it also reported ```"C"```, whereas it reported ```"en_US.UTF-8"``` under my normal Apple Terminal.

What region do you have set in "System Preferences" -> "Language & Region"?
And under the "Advanced..." button, what is your "Format language" set to?

mehdideveloper, 2017-06-13:
I set the LANG environment variable in XQuartz's Terminal (not OS X's Terminal)
Just open XQuartz and then press Cmd+N to open a new xterm window and set LANG.
Editing the "grenv.sh" also solves this problem.
By the way, I don't have "Format Language" in the "Advanced..." section.

[image]

cfriedt, 2017-07-15:
It seems this issue has been solved by the user.

closed 2017-07-15 by cfriedt