ktemkin/gnuradio-for-mac-without-macports #45: Stray symlink for Cheetah.egg?

dholl, 2017-06-11:
Howdy @cfriedt !

While wrapping up #41 (changing all absolute symlinks to relative...), I found an invalid symbolic link being written by ```build.sh```
https://github.com/cfriedt/gnuradio-for-mac-without-macports/blob/master/build.sh#L871

```sh
    P=Cheetah-2.4.4
    URL='https://mirror.csclub.uwaterloo.ca/gentoo-distfiles/distfiles/Cheetah-2.4.4.tar.gz'
    CKSUM=sha256:be308229f0c1e5e5af4f27d7ee06d90bb19e6af3059794e5fd536a6f29a9b550

  LDFLAGS="${LDFLAGS} $(python-config --ldflags)" \
  build_and_install_setup_py \
    ${P} \
    ${URL} \
    ${CKSUM} \
  && ln -sf ${PYTHONPATH}/${P}-py2.7.egg ${PYTHONPATH}/Cheetah.egg
```

Do you recall what the Cheetah.egg symlink is for?  It currently tries pointing to 'Cheetah-2.4.4-py2.7.egg', but the current build process creates a directory called 'Cheetah-2.4.4-py2.7-macosx-10.12-intel.egg' instead.  So my hunch is, may I delete this line from build.sh?  :)  Or should I fix it to point Cheetah.egg to the correct name ```Cheetah-2.4.4-py2.7-macosx-10.12-intel.egg``` --- dynamically figuring this out of course...

```
zero(ttys001):...out-macports> ls -ald /Applications/GNURadio.app/Contents/MacOS/usr/lib/python2.7/site-packages/Cheetah*
drwxr-xr-x  4 dholl  staff  136 Jun  5 20:41 /Applications/GNURadio.app/Contents/MacOS/usr/lib/python2.7/site-packages/Cheetah-2.4.4-py2.7-macosx-10.12-intel.egg
lrwxr-xr-x  1 dholl  staff  109 Jun  9 18:12 /Applications/GNURadio.app/Contents/MacOS/usr/lib/python2.7/site-packages/Cheetah.egg -> /Applications/GNURadio.app/Contents/MacOS/usr/lib/python2.7/site-packages/Cheetah-2.4.4-py2.7.egg
```

cfriedt, 2017-06-24:
Oh weird... that looks like it doesn't belong. I haven't tested it yet though. Have been very busy at work for the last 2 weeks.

cfriedt, 2019-03-18:
If you are able to put together a fix yourself, can demonstrate that the fix works, and make a pull request, I will gladly merge your changes.

However, I no longer have access to a Mac OS machine, so cannot test changes individually. I'm also unable to build new releases unfortunately.

I'd be happy to merge PR's until the time someone else is able to take over maintainership.