ktemkin/gnuradio-for-mac-without-macports #46: Change gr-osmosdr (adding new SDRs)

mehdideveloper, 2017-06-13:
Hi
Is it possible to compile gr-osmosdr for this project and replace with the current version?
You have compiled gr-osmosdr without the nonfree components, so we can't use SDRPlay with it.
I want to be able to use SDRPlay with this project.

Thanks

cfriedt, 2017-06-24:
Hi @mehdideveloper , thanks for the suggestion. You are free to compile it yourself or submit a pull request that will do so. I'm just not sure that SDRPlay is really necessary to the ultimate goal of bundling GNURadio for Mac as an app. If you would like to submit a pull request, and maybe explain a bit more about the nonfree components, I would be happy to take a look to see if we can incorporate it.

cfriedt, 2017-07-14:
I'm taking a look at this now.

mehdideveloper, 2017-07-14:
@cfriedt Can I help?

cfriedt, 2017-07-14:
Should be no problem. Keep an eye out for updates on this page to see progress.

cfriedt, 2017-07-15:
Hmm... this warning seems prohibitive
```
-- NONFREE components have been enabled. The resulting
   binaries cannot be distributed under GPL terms.
```

cfriedt, 2017-07-15:
That's a bit of a show-stopper. In order to continue distributing this under the GPL, I'm going to leave it up to end users to recompile on their own version of gr-osmosdr in case they need it.

Sorry!

closed 2017-07-15 by cfriedt

cfriedt, 2017-07-16:
I've left a [branch](https://github.com/cfriedt/gnuradio-for-mac-without-macports/tree/issue/46-change-gr-osmosdr-adding-new-sdrs) open so that users can easily pull it in and build themselves.

Suggest pre-populating tmp/.[foo].done with all packages except gr-osmosdr so that only gr-osmosdr needs to be rebuilt