ktemkin/gnuradio-for-mac-without-macports #58: building problem related with pkg-config-0.29.1

pcwhy, 2017-08-20:
When building in MAC OS X 10.12.6 I got the following error..

D: already downloaded pkg-config-0.29.1.tar.gz
D: File '/Users/liuyongxin/gnuradio-for-mac-without-macports-master/tmp/pkg-config-0.29.1.tar.gz' matches 'beb43c9e064555469bd4390dcfd8030b1536e0aa103f08d7abf7ae8cac0cb001'
I: Extracting pkg-config-0.29.1.tar.gz to pkg-config-0.29.1
tar xpzf pkg-config-0.29.1.tar.gz
I: Running autoreconf in pkg-config-0.29.1
m4macros/glib-gettext.m4:39: error: m4_copy: won't overwrite defined macro: glib_DEFUN
m4macros/glib-gettext.m4:39: the top level
autom4te: /usr/bin/m4 failed with exit status: 1
aclocal: error: echo failed with exit status: 1
autoreconf: aclocal failed with exit status: 1
E: autoreconf failed for pkg-config-0.29.1

and any solution to mitigate that?

cfriedt, 2017-08-31:
None as of yet. It's a problem already reported upstream though
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=822409

cfriedt, 2019-03-18:
If you are able to put together a fix yourself, can demonstrate that the fix works, and make a pull request, I will gladly merge your changes.

However, I no longer have access to a Mac OS machine, so cannot test changes individually. I'm also unable to build new releases unfortunately.

I'd be happy to merge PR's until the time someone else is able to take over maintainership.