ktemkin/gnuradio-for-mac-without-macports #59: GNURadio-v3.7.10.1-rc2.dmg

Hardcore-fs, 2017-09-24:
Hi,

loading it , quarts starts up, then after about a minute the "apple script editor" window opens

then nothing.

cfriedt, 2017-09-24:
Did you install Python 2.7 and XQuartz?

Hardcore-fs, 2017-09-24:
yep......
something strange,  if i open a terminal
then go to the "notification window" and click on the message so it opens again, after a minute  the program starts.

so that would prove python &  XQuartz are installed.

I'm going to take a look today, maybe the program file is incorrectly associated with the script editor

Hardcore-fs, 2017-09-26:
yes seems something minor is not initting.
Just over wrote the "app" and exactly the same thing happened the first time it is launched.
After the first run it does not do it again, until you overwrite the app again.

cfriedt, 2017-09-27:
@Hardcore-fs Python 2.7 specifically? GNURadio does not (yet) work with Python 3

cfriedt, 2017-09-27:
That's weird... I'll try to reproduce what you're seeing on my end too, and if I can, I'll create a branch to fix this issue.

Hardcore-fs, 2017-09-27:
python 2.7 is in and being used, the app only does it the first time
osx 10.11.6  16GB ram
also the path is:
 ". /Applications/GNURadio.app/Contents/MacOS/usr/bin/grenv.sh "

Also when it is launching for the first time:
is says   "launching gnuradio-companion -psn_0_421991"

but all runs after the first run it says "launching gnuradio-companion"
[image]

cfriedt, 2017-09-28:
Interesting. I wonder if @dholl could comment, since he introduced the code for the script windows. I haven't yet had a chance to reproduce it myself.

dholl, 2017-09-29:
@Hardcore-fs Hmmm....  I wasn't able to reproduce (yet) either, but I'm tracing through the logic of the launch script "run-grc.sh"...

As a hack test, could you edit this script: ```GNURadio.app/Contents/MacOS/usr/bin/run-grc```
and make these changes:
1. Uncomment this block near the end:
    ```sh
    ## Strip out the -psn_... argument added by the OS.
    #if [ "x${1#-psn_}" != "x${1}" ]; then
    #	shift 1
    #fi
    ```

2. And move that block before this ```if``` statement
    ```sh
    if command -v xset >/dev/null 2>&1 ; then
        # If xset is available,
        ...
    else
        ...
    fi
    ```

The complete script should look like this:
```sh
#!/bin/sh

XQUARTZ_APP_DIR=/Applications/Utilities/XQuartz.app
PYTHON_FRAMEWORK_DIR=/System/Library/Frameworks/Python.framework/Versions/2.7

set -e

show_fail_message () {
	osascript -e 'display notification "Failed to start GNU Radio Companion.\nDetail: '"$*"'" with title "GNU Radio Companion"' > /dev/null 2>&1 || true
	return 0
}

# If the exec fails for some reason, then our EXIT handler will get invoked.
# At exit, run cleanup_script but let the exit status still be whatever the
# shell was going to exit with:
trap 'show_fail_message "exit status $?"' EXIT

# If we get a signal such as SIGINT SIGQUIT or SIGTERM, then mask signal,
# show_fail_message, restore default handler, and raise signal again to let it
# propagate to whatever process started this script:
trap '{ trap - EXIT ; trap "" INT  ; show_fail_message "received signal INT"  ; trap - INT  ; kill -s INT  $$ ; }' INT
trap '{ trap - EXIT ; trap "" QUIT ; show_fail_message "received signal QUIT" ; trap - QUIT ; kill -s QUIT $$ ; }' QUIT
trap '{ trap - EXIT ; trap "" TERM ; show_fail_message "received signal TERM" ; trap - TERM ; kill -s TERM $$ ; }' TERM
# The leading "trap - EXIT" disables the EXIT signal handler or otherwise
# show_fail_message gets called twice.

# Uncomment one at a time to test each exit condition:
#kill -s INT -- $$
#kill -s QUIT -- $$
#kill -s TERM -- $$
#exit 2

# Figure out where this script is located:
case "$0" in
	/*)
		# $0 is absolute path since it starts with /
		argv0_path="$0"
		;;
	*)
		# Assume $0 is relative path:
		argv0_path="$PWD/$0"
		;;
esac
# This script is in GNURadio.app/Contents/MacOS/usr/bin with grenv.sh
grenv_path="${argv0_path%/*}/grenv.sh"

if ! test -d ${XQUARTZ_APP_DIR} ; then
    osascript \
        -e 'on run(argv)' \
        -e 'display dialog ("XQuartz is not installed. Download it at http://www.xquartz.org/") buttons {"OK"} default button 1 with icon stop with title "GNU Radio Companion"' \
        -e 'end run' \
        > /dev/null 2>&1 || true
    printf 'XQuartz is not installed. Download it at http://www.xquartz.org/\n' 1>&2
fi

if ! test -d ${PYTHON_FRAMEWORK_DIR} ; then
    osascript \
        -e 'on run(argv)' \
        -e 'display dialog ("Python 2.7 is not installed. Download it here: https://www.python.org/downloads/") buttons {"OK"} default button 1 with icon stop with title "GNU Radio Companion"' \
        -e 'end run' \
        > /dev/null 2>&1 || true
    printf 'Python 2.7 is not installed. Download it here: https://www.python.org/downloads/\n' 1>&2
fi

if ! test -e "${grenv_path}" ; then
	osascript \
		-e 'on run(argv)' \
		-e 'display dialog ("Unable to find grenv.sh at " & item 1 of argv) buttons {"OK"} default button 1 with icon stop with title "GNU Radio Companion"' \
		-e 'end run' \
		-- "${grenv_path}" \
		> /dev/null 2>&1 || true
	printf 'Unable to find grenv.sh at %s\n' "${grenv_path}" 1>&2
	exit 1
fi
. "${grenv_path}"

# Strip out the -psn_... argument added by the OS.
if [ "x${1#-psn_}" != "x${1}" ]; then
	shift 1
fi

if command -v xset >/dev/null 2>&1 ; then
	# If xset is available, then we'll use that to silently launch the X server.
	# While we wait, use osascript to tell the user what we're up to.
	osascript -e 'display notification "Starting X server..." with title "GNU Radio Companion"' > /dev/null 2>&1 || true
	if xset q >/dev/null 2>&1 ; then
		osascript \
			-e 'on run(argv)' \
			-e 'display notification ("exec gnuradio-companion " & item 1 of argv) with title "GNU Radio Companion" subtitle "Launching GNU Radio Companion..."' \
			-e 'end run' \
			-- "$*" \
			> /dev/null 2>&1 || true
	else
		osascript -e 'display notification "Unable to verify running X server.  Will attempt anyway..." with title "GNU Radio Companion"' > /dev/null 2>&1 || true
	fi
else
	osascript \
		-e 'on run(argv)' \
		-e 'display notification ("exec gnuradio-companion " & item 1 of argv) with title "GNU Radio Companion" subtitle "Launching GNU Radio Companion..."' \
		-e 'end run' \
		-- "$*" \
		> /dev/null 2>&1 || true
fi

# Prime the exit status with "( exit 127 ) || " in case exec fails.  Otherwise,
# $? is 0 during trap EXIT.
( exit 127 ) || exec gnuradio-companion "${@}"
# Without the "trap ... EXIT", then we wouldn't need the "( exit 127 ) || ",
# since the shell would automatically return the appropriate error number from
# exec's failure.  Is this a glitch in bash since sh is symlinked to bash on my
# system?  Note that exec can return other exit status beside 127, so this
# stunt is suboptimal, but an optimally pedantic solution would require exec to
# interact better wirth trap EXIT.
```

I just attached a copy.  Rename it from run-grc.txt to run-grc, and ensure its file permissions are 755 such as ```chmod 755 /Applications/GNURadio.app/Contents/MacOS/usr/bin/run-grc```
[run-grc.txt](https://github.com/cfriedt/gnuradio-for-mac-without-macports/files/1343041/run-grc.txt)

Hardcore-fs, 2017-09-30:
Yep.... i'm on an old early 2008 tank (10.11.6) so no further updates are allowed by Apple
Not to be pedantic.......
Surely the "correct" instruction would be:
1. download a clean copy of "GNURadio.app" (to trigger bug, since it only does it first time)
2. replace old script in "path" with enclosed script
3. test

That said....... two tests.
1. re-install unmodified package (broken on initial launch)

2.  re-Install AND modify/replace script,  on running application problem appears solved.

[image]

dholl, 2017-10-01:
huh....  So, the tweaked script fixes the issue?  I wonder if there's something about that -psn#### arg being passed to osascript for the "exec ..." diagnostic message, or if it is python / gtk trying to do something with that argument when grc starts.

I just submitted a PR.

Hardcore-fs, 2017-10-01:
Ahh.. the mystery that is computer code.
like I say,  it only happens the first time the program is cleanly installed, so most people will put it down to " oh, ill just restart it again."

But it is reproducible(on my machine) and the fix is reproducible on my machine, so all is good
but we can see from above , on the 1st launch with the script fix,  no "-psn### was used. (1m)

and even with the "broken" script(10m), that "-psn###" appears to only be used the first time and not subsequently.

Anyway you guys did a good job, its the only startup issue i have had, all other issues are internal module related, so  ill report them to GNU

dholl, 2017-10-02:
Thank you!

Yeah, one source of start-up mysteries is that GRC does not bother to process native [Apple Events](https://developer.apple.com/library/content/documentation/Cocoa/Conceptual/ScriptableCocoaApplications/SApps_handle_AEs/SAppsHandleAEs.html), because it is pyGTK and [handling native Mac events in pyGTK seems to require more think-time than I have available at the moment](https://wiki.gnome.org/Projects/GTK+/OSX/Integration).  If anyone is game for it, this integration with Finder / Launch Services might be a nifty opportunity to contribute to @cfriedt's work.  :)

-- aside:

Otherwise, I have been slowly toying around with an experimental build that does not require X11 but does leverage MacPorts.  I've only recently gotten the .app bundle to be relocatable, and still need to get around to testing Launch Services / Finder integration...  However, in this alternative build, I haven't focused at all on the extra packages / options that @cfriedt has included in this repo.  [I just enable whatever bells-n-whistles are available in MacPorts, and then belt-sand the whole thing down to hide a trimmed-down MacPorts tree inside the .app bundle.]  :)

Hardcore-fs, 2017-10-02:
As a developer ,I would stay away from mac ports and all these other package /build shells.
The cleanliness of "X11" is where it's at.
perhaps that is my issue with this package failing : 10 years of past crap , mac ports, java libraries , x11, arm tools, python 2.7 & 3.0
Just too many interactions.

I think I have seen "issues" with the  mac events, both on the sliders & source/ sink packages in GNC

like the sound driver being able to block events is potentially "dangerous" , plus having seen sliders track from 0-100 with steps of 2 , suddenly jumping to 900K or 900M

cfriedt, 2017-10-03:
@Hardcore-fs - I completely hear you. My next GNU Radio project will likely be a native Mac OS X UI using Cocoa. Next... maybe a browser-based UI for Chromium / Chrome ? I only ever switched to Mac because the Linux desktop started spiraling.

Hardcore-fs, 2017-10-03:
Actually that might work!!!!!
a backend CLI (possibly GNUradio with  API) and a web front end, get away from all this  cross platform" nastiness"
"Radare"  is making good inroads into this with their disassembler.

I would drop the "mac specific" app Idea, I regularly work on multiple platforms switching between them, so a web interface would really work, plus there is potentially a money maker in there....
from an educational/share designs/ "cloud" point of view.

cfriedt, 2017-10-04:
@Hardcore-fs , I'm tempted to use the Electron framework that @github uses themselves.

zendata, 2017-12-02:
[image]

Any idea why I would get this display? It appears that the receiver is running OK but QT is not rendering properly. OS 10.13.1 on MacBook Air.

cfriedt, 2017-12-02:
@zendata Are you able to provide your .grc file?

zendata, 2017-12-03:
> Done

Hi Chris

Sure I attach the .grc file, the .py file and the run notes.

Thanks for having a look - happy to help you test - the idea of a native Mac app version of gnuradio is so appealing, as someone who has struggled for ever with MacPorts implementations and Linux VMs with their USB port issues etc.

Cheers,
Graham Cottew
Technology Director
Zendata Pty Ltd
+61417722400

Run notes:
Generating: '/Users/graham/Dropbox/Gnuradio/cfriedt_dmg_test.py'

Executing: /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python -u /Users/graham/Dropbox/Gnuradio/cfriedt_dmg_test.py

gr-osmosdr v0.1.4-98-gc653754d (0.1.5git) gnuradio 3.7.10.1
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya 
[32;1m[INFO] [UHDMac OS; Clang version 8.1.0 (clang-802.0.42); Boost_106300; UHD_3.11.0.local-master-233-g25fc32af] [39;0m
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
[R82XX] PLL not locked!

nsmith-, 2018-02-14:
Hi @zendata,

I faced a similar issue and found the problem was that fontconfig was not searching the correct directories on my system, for whatever reason.  The solution was to edit `/Applications/GNURadio.app/Contents/MacOS/usr/etc/fonts/fonts.conf` and change the nonexistent directory on line 27:
```
	<dir>/usr/X11R6/lib/X11/fonts</dir>
```
to
```
	<dir>/usr/X11/lib/X11/fonts</dir>
```
which seems to be a symlink to my xquartz installation in `/opt/X11`

P.S. Actually this was the latter half of the struggle.  The first pain was learning that it does not work in El Capitan and (about time) upgrading to Sierra.  This was because fontconfig would segfault if it tried to use a syscall that doesn't exist in MacOS until Sierra, namely `mkostemp` More info [here](https://bugs.freedesktop.org/show_bug.cgi?id=102986)
Perhaps it would be good to add the flag mentioned to the fontconfig build step.

zendata, 2018-02-15:
Hi Nicholas 

Thanks your help. 
I will try your suggested change - I am running High Sierra and guess that may bring another few issues over Sierra. 

Will let you know when I have had a go at it. 

Cheers,
Graham

cfriedt, 2018-02-23:
I did finally upgrade to High Sierra (reluctantly, because I was losing RAID support), so now I should be able to give this a shot soonish.

Vladivostoks, 2018-06-12:
hi, i try to open it with run-grc ,but there is something like "ImportError: No module named grc.main".i can import gnuradio module in python2 console, but it seems grc.main is missing?Could anyone help me?

discort, 2019-03-17:
still have the same issue,
release [v3.7.10.1-rc2](https://github.com/cfriedt/gnuradio-for-mac-without-macports/tree/v3.7.10.1-rc2)

cfriedt, 2019-03-18:
If you are able to put together a fix yourself, can demonstrate that the fix works, and make a pull request, I will gladly merge your changes.

However, I no longer have access to a Mac OS machine, so cannot test changes individually. I'm also unable to build new releases unfortunately.

I'd be happy to merge PR's until the time someone else is able to take over maintainership.

ernestp, 2019-10-17:
Having same font issue with uhd_fft.grc from gnuradio github sources

cfriedt, 2019-10-17:
Thanks, @ernestp - I'm not sure if @ktemkin will be maintaining a 3.7 release branch, but this will likely be fixed in 3.8. Currently, there is a 3.8 pre-release.

ernestp, 2019-10-18:
@cfriedt @ktemkin It seems 3.8 currently have a lot of issues with UHD. Maintaining 3.7 will be good idea.

cfriedt, 2019-10-20:
@ernestp can you please open a separate issue for whatever you're having? This one seems to cover several unrelated issues.

ernestp, 2019-10-21:
@cfriedt Sure, already did https://github.com/gnuradio/gnuradio/issues/2851