ktemkin/gnuradio-for-mac-without-macports #61: Latest Release Does not Launch

nkizz, 2017-12-05:
Hi there,
I just downloaded the latest release (v3.7.10.1-rc2), and it's not launching on my Sierra machine. I get two notifications, "Launching X Server" and "Launching GNU Radio Companion", then nothing. Running "/Applications/GNURadio.app/Contents/MacOS/usr/bin/run_grc" gives this error:
`Traceback (most recent call last):
  File "/Applications/GNURadio.app/Contents/MacOS/usr/bin/gnuradio-companion", line 96, in <module>
    check_gnuradio_import()
  File "/Applications/GNURadio.app/Contents/MacOS/usr/bin/gnuradio-companion", line 70, in check_gnuradio_import
    from gnuradio import gr
  File "/usr/local/lib/python2.7/site-packages/gnuradio/gr/__init__.py", line 37, in <module>

  File "/usr/local/lib/python2.7/site-packages/gnuradio/gr/runtime_swig.py", line 1411, in <module>
    __swig_destroy__ = _runtime_swig.delete_gr_vector_vector_complexd
AttributeError: 'module' object has no attribute 'sizeof_char_swigconstant'`
I've tried Googling it to no avail. I'm not sure if this is an upstream error or not. It could also be because I have various packages installed from brew.
Any help is appreciated!

cfriedt, 2019-03-02:
Looks like a DYLD_LIBRARY_PATH issue, or maybe you're using the wrong python.

What does `which python` say when you execute it from the command line?

We definitely should not be referencing anything in /usr/local/lib/python2.7 . Did you install python from here?

https://www.python.org/downloads/release/python-2715/

cfriedt, 2019-03-18:
If you are able to put together a fix yourself, can demonstrate that the fix works, and make a pull request, I will gladly merge your changes.

However, I no longer have access to a Mac OS machine, so cannot test changes individually. I'm also unable to build new releases unfortunately.

I'd be happy to merge PR's until the time someone else is able to take over maintainership.

beaugunderson, 2020-05-21:
closing this as there's a new release that seems to be working well for most people now--feel free to reopen if this same stacktrace shows up again :)

closed 2020-05-21 by beaugunderson