ktemkin/gnuradio-for-mac-without-macports #62: How can I install OOT modules?

misll75, 2017-12-17:
Hello!

First of all.... Thanks a lot for your work here!
Now I can run Gnuradio on my mac like a charm. :-)

But there is one question... How can I build/install OOT modules into gr now?

Best regards!
Andi

cfriedt, 2017-12-20:
That's a good question! I have not got to that part in terms of adding it to the README.md file yet.

It should be very easy, and I would welcome a contribution :-)

In theory, everything you need to build your own apps lives under /Applications/GNURadio.app/Contents/MacOS, with the only exception being Apple's Command-Line developer utilities (which you should be prompted to install if you type `gcc` or `xcode-select --install`).

So the first thing you would likely want to do is source the grenv.sh script, which will set up all of your necessary paths.

```
. /Applications/GNURadio.app/Contents/MacOS/usr/bin/grenv.sh
```

Then, follow the instructions at https://wiki.gnuradio.org/index.php/OutOfTreeModules .

Of course, I'm just pointing you in the right direction. If you were able to document all of the procedures you took and wrote a step-by-step list, I would very happily add it to README.md !

cfriedt, 2019-03-18:
If you are able to put together a fix yourself, can demonstrate that the fix works, and make a pull request, I will gladly merge your changes.

However, I no longer have access to a Mac OS machine, so cannot test changes individually. I'm also unable to build new releases unfortunately.

I'd be happy to merge PR's until the time someone else is able to take over maintainership.