ktemkin/gnuradio-for-mac-without-macports #63: python libraries not installed system-wide

calebmadrigal, 2018-02-14:
I'm not sure if this is a bug, or a feature, or if I installed it incorrectly, but the python libraries in `/Applications/GNURadio.app/Contents/MacOS/usr/lib/python2.7/site-packages` didn't get copied or linked to `/usr/lib/python2.7/site-packages` for me (High Sierra).

In the interim, I'll probably just update my python path.

nsmith-, 2018-02-15:
I think the idea is that everything is self-contained inside the app bundle, and it is pre-built against the system python which is at least for me (Sierra) not in `/usr/lib`.  Regarding the latter point, this works for me:
```bash
# setup environment
source /Applications/GNURadio.app/Contents/MacOS/usr/bin/grenv.sh
# perhaps an alias would help here
/System/Library/Frameworks/Python.framework/Resources/Python.app/Contents/MacOS/Python top_block.py
```

cfriedt, 2018-02-23:
@nsmith- is correct.

closed 2018-02-23 by cfriedt