ktemkin/gnuradio-for-mac-without-macports #64: Will there be 3.7.11 any time ?

helmarw, 2018-03-05:
just wondering if this repository is still being maintained?!
I like it a lot, since i had some trouble with homebrew, may thnx !!!

Will there be any updates e.g. to version 3.7.11 ?
thnx
Helmar

gastonet, 2018-03-15:
+1 Please.

cfriedt, 2018-04-26:
Hey guys!

For positive feedback like that, I'll see what I can do ;-) Note, it's pretty easy to build yourself. I typically use a Mac OS X VM to do the actual building to sandbox things.

Will get a build going shortly and let you know..

cfriedt, 2019-03-18:
If you are able to put together a fix yourself, can demonstrate that the fix works, and make a pull request, I will gladly merge your changes.

However, I no longer have access to a Mac OS machine, so cannot test changes individually. I'm also unable to build new releases unfortunately.

I'd be happy to merge PR's until the time someone else is able to take over maintainership.

ktemkin, 2019-10-09:
There's a 3.8.0.0, including a pre-release; which is hopefully ~0.889 better. ;)

Closing this issue, as things are hopefully being maintained again.

closed 2019-10-09 by ktemkin