ktemkin/gnuradio-for-mac-without-macports #65: Build/downoad failure for orc-0.4.26 and lxml-3.7.3

dl9rdz, 2018-03-13:
Using https://gstreamer.freedesktop.org/src/orc/orc-0.4.26.tar.xz
instead of https://mirror.csclub.uwaterloo.ca/gentoo-distfiles/distfiles/orc-0.4.26.tar.xz
fixes the problem. The latter URL returns a 404 error

same problem with lxml-3.7.3.tar.gz, can be fixed by replacing the uwaterloo mirror with
https://pypi.python.org/packages/39/e8/a8e0b1fa65dd021d48fe21464f71783655f39a41f218293c1c590d54eb82/lxml-3.7.3.tar.gz#md5=075692ce442e69bbd604d44e21c02753

Hansis-Air:gnuradio-for-mac-without-macports hansi$ ./build.sh 
I: BUILD_DIR = '/Users/hansi/src/gnuradio-for-mac-without-macports'
I: INSTALL_DIR = '/Applications/GNURadio.app/Contents/MacOS'
I: already installed autoconf-2.69
I: already installed automake-1.15
I: already installed libtool-2.4
I: already installed gettext-0.19.8
I: already installed xz-5.2.3
I: already installed tar-1.29
I: already installed pkg-config-0.29.1
I: already installed pcre-8.40
I: already installed swig-3.0.12
I: already installed libffi-3.2.1
I: already installed glib-2.51.1
I: already installed cppunit-1.12.1
I: already installed Mako-1.0.3
I: already installed bison-3.0.4
I: already installed openssl-1.1.0d
I: already installed thrift-0.10.0
I: fetching orc-0.4.26 from https://mirror.csclub.uwaterloo.ca/gentoo-distfiles/distfiles/orc-0.4.26.tar.xz
D: checking for /Users/hansi/src/gnuradio-for-mac-without-macports/tmp/orc-0.4.26.zip
D: checking for /Users/hansi/src/gnuradio-for-mac-without-macports/tmp/orc-0.4.26.tar.gz
D: checking for /Users/hansi/src/gnuradio-for-mac-without-macports/tmp/orc-0.4.26.tgz
D: checking for /Users/hansi/src/gnuradio-for-mac-without-macports/tmp/orc-0.4.26.tar.bz2
D: checking for /Users/hansi/src/gnuradio-for-mac-without-macports/tmp/orc-0.4.26.tbz2
D: checking for /Users/hansi/src/gnuradio-for-mac-without-macports/tmp/orc-0.4.26.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   178  100   178    0     0    307      0 --:--:-- --:--:-- --:--:--   308
E: File 'orc-0.4.26.tar.xz' does not match '7d52fa80ef84988359c3434e1eea302d077a08987abdde6905678ebcad4fa649'
Hansis-Air:gnuradio-for-mac-without-macports hansi$

cfriedt, 2018-05-17:
Thanks! Feel free to make a pull request :-)

cfriedt, 2019-03-18:
If you are able to put together a fix yourself, can demonstrate that the fix works, and make a pull request, I will gladly merge your changes.

However, I no longer have access to a Mac OS machine, so cannot test changes individually. I'm also unable to build new releases unfortunately.

I'd be happy to merge PR's until the time someone else is able to take over maintainership.

ktemkin, 2019-10-16:
Fixed in the gr3.8 build. Closing.

closed 2019-10-16 by ktemkin