ktemkin/gnuradio-for-mac-without-macports #66: Bison build error

jimla44, 2018-06-12:
Thank you very much for providing the script for building GNU Radio for MacOS. I ran the script using MacOS High Sierra 10.13.5 and got the following error message:

  CCLD     src/bison
  GEN      doc/bison.help
/bin/sh: line 1: 59990 Abort trap: 6           src/bison --version > doc/bison.help.tmp
make[2]: *** [doc/bison.help] Error 134
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
E: failed to configure, make, and install bison-3.0.4

cfriedt, 2018-06-26:
Hi there - rather than running it in your 'production' environment, I would suggest running it inside of a VM. I use VMWare Fusion. The main reason, is to have a clean & consistent build environment. I can't remember whether I've tried to build it using High Sierra, but it definitely worked with Sierra.

cfriedt, 2019-03-18:
If you are able to put together a fix yourself, can demonstrate that the fix works, and make a pull request, I will gladly merge your changes.

However, I no longer have access to a Mac OS machine, so cannot test changes individually. I'm also unable to build new releases unfortunately.

I'd be happy to merge PR's until the time someone else is able to take over maintainership.

ktemkin, 2019-10-16:
Fixed in the gr3.8 build. Closing.

closed 2019-10-16 by ktemkin