ktemkin/gnuradio-for-mac-without-macports #68: Unable to Verfify X Server after Installation on Startup

dev7ch, 2018-12-03:
Installed on Mac Mojave.

cfriedt, 2018-12-04:
Any additional information would be helpful. Screenshot maybe? Try running it from the command line?

dev7ch, 2018-12-05:
Sorry, yes forgot to post the rest of this issue:
- yes i tried running from command line with `startx radio-companion` in iTerm terminal
- I realized that i might be the missing X Server from Apple 

I assume, coz X Server from isn`t installed it dont work.

Is X Server a depende3cy as well? which have to be bought? (it`s not free as far i see)

ekiojp, 2018-12-26:
This may work,

* Install XQuartz from https://www.xquartz.org/releases/index.html (2.7.11)
* Start App XQuartz
* Open a Terminal inside XQuartz (Applications -> Terminal)
* Run /Applications/GNURadio.app/Contents/MacOS/usr/bin/run-grc (inside the XQuartz new terminal)

cfriedt, 2019-03-18:
If you are able to put together a fix yourself, can demonstrate that the fix works, and make a pull request, I will gladly merge your changes.

However, I no longer have access to a Mac OS machine, so cannot test changes individually. I'm also unable to build new releases unfortunately.

I'd be happy to merge PR's until the time someone else is able to take over maintainership.

dev7ch, 2019-03-18:
thx for the update

Elfsong, 2020-07-19:
> This may work,
> 
> * Install XQuartz from https://www.xquartz.org/releases/index.html (2.7.11)
> * Start App XQuartz
> * Open a Terminal inside XQuartz (Applications -> Terminal)
> * Run /Applications/GNURadio.app/Contents/MacOS/usr/bin/run-grc (inside the XQuartz new terminal)

Thank you so much. It works!