ktemkin/gnuradio-for-mac-without-macports #69: Missing fonts

tfblum, 2019-02-26:
It appears that some required fonts are missing. 
OS: Mac OS 10.14.3

[image]

cfriedt, 2019-03-02:
Odd... sorry, wish I could try and replecate your problem, but I've kind of given up on Mac OS X, well... because Apple gave up on my MacbookPro8,3 and I don't feel like throwing another $2k down the drain.

You could always try building it from scratch ¯\_(ツ)_/¯

cfriedt, 2019-03-18:
If you are able to put together a fix yourself, can demonstrate that the fix works, and make a pull request, I will gladly merge your changes.

However, I no longer have access to a Mac OS machine, so cannot test changes individually. I'm also unable to build new releases unfortunately.

I'd be happy to merge PR's until the time someone else is able to take over maintainership.

tfblum, 2019-03-18:
Hi Christopher,

Thanks for your advice. I will certainly let you know if can fix it. For now I am quite busy with other projects but I when I get a chance to try and compile it myself I will let you know how it goes.

Best,

Thomas Blum

ktemkin, 2019-10-09:
If this is still relevant, could you try one of the 3.8.0.0-series pre-releases, and let me know if you're still having font issues?

b3n01t, 2019-10-16:
I had the same issue. I gave 3.8.0.0 a shot. It did not work :/ 
Looks like I need to install python 3. I will eventually do.

On my system, the issue is that the font files  are not at the expected location.
In `/Applications/GNURadio.app/Contents/MacOS/usr/etc/fonts/fonts.conf`,  you can find this.:

```xml
<!-- Font directory list -->
    <dir>/usr/share/fonts</dir>
    <dir>/usr/X11R6/lib/X11/fonts</dir>
    <dir prefix="xdg">fonts</dir>
    <!-- the following element will be removed in the future -->
    <dir>~/.fonts</dir>
```

None of this directory exist on my system. But XQuartz provides `/opt/X11/share/fonts`. And a simple 
```
> ln -s /opt/X11/share/fonts ~/.fonts
```
ended up fixing the issue for me.

ktemkin, 2019-10-16:
Interesting -- generally, the path `/usr/X11R6` is configured to be effectively a link to `/opt/X11`. This occurs during XQuartz install, but apparently this is undone when major macOS updates happen.

One can re-create the relevant link with:

```
sudo /usr/libexec/x11-select /opt/X11
```

However, looking at this, I'm considering just adding `/opt/X11/lib/X11/fonts` as a fontconfig font path, so this will work whether or not that link is present.

nsbruce, 2020-08-24:
The 3.8 release doesn't have the fonts problem for me but the 3.7 does and @b3n01t 's solution fixed it.