ktemkin/gnuradio-for-mac-without-macports #71: GRC shows block groups, but no blocks.

playaspec, 2019-10-13:
GNURadio.app requires the "official" Python 3.7 package to run, but official package doesn't include all dependencies required for GRC to run correctly. Running GRC from the XQuartz terminal yields:

```
bash-3.2$ /Applications/GNURadio.app/Contents/MacOS/usr/bin/run-grc 
WARNING:gnuradio.grc.core.platform:Slow YAML loading (libyaml not available)
<<< Welcome to GNU Radio Companion 3.8.0.0 >>>

Block paths:
        /Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/grc/blocks
```

After some investigation I found the Python.app's resources in /Library/Frameworks/Python.framework//Versions/3.7/bin/. Pip is not installed by default. I installed pip, and ran "/Library/Frameworks/Python.framework//Versions/3.7/bin/pip install pyyaml" which installed PyYAML. Running: 

```
$  /Library/Frameworks/Python.framework//Versions/3.7/bin/python3.7
Python 3.7.4 (v3.7.4:e09359112e, Jul  8 2019, 14:54:52)
[Clang 6.0 (clang-600.0.57)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import yaml
>>> print (yaml.__with_libyaml__)
False
```
Shows that the Python.app now has the yaml library, but it is not libyaml enabled.

As a consequence, GRC shows block _categories_, but no blocks.

[image]

As you can see there are no arrows to expand, and no blocks available. FWIW I initially assumed that the GNURadio.app would find my python3.7 install from Macports, which has a working libyaml based yaml library, and is selected as the default. Perhaps GRC the first time tried to set up the environment on the first run while Python.app was missing and is now relying on stale settings?

I am at a loss on how to enable a libyaml based yaml library for the official Python.app. Perhaps the missing blocks are unrelated to the missing libyaml support, but the error from run-grc  was the only thread I could pull on.

ktemkin, 2019-10-16:
There's a known bug in this 'beta' version in which a few of the icons needed for GTK3 UI didn't wind up installed. As a result, 'tree view' elements don't properly have the "open/closed" icons next to them. (This will be fixed in the next build.)

This is a little bit silly, but can you try each of the three following things?

- Click _just to the left_ of the category text, as if there were an icon, and see if the category expands?
- Click on the category text, and once it's selected, press Return/Enter?
- Click inside the list, and then press CTRL+F (_not_ Cmd+F), and then search for the name of a block ("frequency" will turn up a few results).

GNURadio populates the list of categories from the YAML that describes the blocks; so it'd be very unusual for it to see categories but not blocks.

> Shows that the Python.app now has the yaml library, but it is not libyaml enabled.

This is correct -- `pyyaml` ships with both a native-code and a pure-python YAML parser. GRC barely does any YAML parsing, so I've skipped shipping libyaml in the app distribution. This shouldn't affect functionality at all.

> I am at a loss on how to enable a libyaml based yaml library for the official Python.app. 

The 'official' Python distribution isn't actually providing our python dependencies -- rather, when we run GRC, we set up the environment's `PYTHONPATH` to include dependencies from inside the .app. The `pyyaml` instance we use is actually located in `<GNUradio.app>/Contents/MacOS/usr/lib/python3.7/site-packages`.

> FWIW I initially assumed that the GNURadio.app would find my python3.7 install from Macports

Unfortunately, a bunch of things link against `libpython`, so in order to have a single distributable set of binaries, we also need to have a fixed distribution of python we're linking against . [In the future, it might make more sense for us to ship a `python3` distribution inside the app; but that's unrelated to this. :)]

playaspec, 2019-10-18:
On Tue, Oct 15, 2019 at 10:52 PM Kate Temkin <notifications@github.com>
wrote:

> This is a little bit silly, but can you try each of the three following
> things?
>
>    - Click *just to the left* of the category text, as if there were an
>    icon, and see if the category expands?
>
> This did indeed work! I initially thought the triangle was left justified
close to the frame divider, and explains why my previous attempts to expand
a category failed.

>
>    - Click on the category text, and once it's selected, press
>    Return/Enter?
>
>
Also worked. I didn't think to try this before.

>
>    - Click inside the list, and then press CTRL+F (*not* Cmd+F), and then
>    search for the name of a block ("frequency" will turn up a few results).
>
> Works as expected.

> GNURadio populates the list of categories from the YAML that describes the
> blocks; so it'd be very unusual for it to see categories but not blocks.
>
> Shows that the Python.app now has the yaml [...]
Great. Then I won't pursue installing it.

Thanks for the help!