ktemkin/gnuradio-for-mac-without-macports #78: Error loading libsoapySupport.so when running flowgraphs

beaugunderson, 2020-03-27:
This doesn't actually affect me since I'm not using soapy, but I see this every time I run my flowgraph:

```
Error: EnvironmentError: OSError: dlopen failed to load "/Applications/GNURadio.app/Contents/MacOS/usr/lib/uhd/modules/libsoapySupport.so"
```

macOS 10.15.4 using v3.8.0.0-pre1.

ktemkin, 2020-03-28:
I think this will be fixed in the next rebuild -- I'll have to check once I have time to run a long-running build. :)

beaugunderson, 2020-03-28:
Nice! Speaking of long-running builds, I played around with setting up AppVeyor for CI; I think build time might be too long though... My [last attempt](https://github.com/ktemkin/gnuradio-for-mac-without-macports/compare/master...beaugunderson:bg-appveyor) died at 50 minutes because one of the mirrors didn't have the right version anymore but AppVeyor caps builds at 60 minutes.