ktemkin/gnuradio-for-mac-without-macports #81: Set up CI builds

ktemkin, 2020-04-05:
Continuing discussion and work from #80. We want to set up a CI service for automatic builds.

It looks like most of this work has already been done; but I need to communicate with @beaugunderson to get our CI service actually set up. :)

beaugunderson, 2020-04-05:
With #80 merged all that needs to be done is to sign up for CircleCI and authorize this repo (and then add payment information). I'm happy to do the first part, I'd need maintainer access to do so. :)

ktemkin, 2020-04-08:
Giving this a try -- there's a build running; let's hope it works. :)

In the future, it might make sense to set up a second branch for CI builds. We could easily fast-forward it to match master each time we want to kick off a build. Thoughts?

beaugunderson, 2020-04-08:
Awesome!

There's the "hold workflow" to require manual approval for builds:

https://circleci.com/docs/2.0/workflows/#holding-a-workflow-for-a-manual-approval

That would allow any commit to be built but only if someone clicks the button in Circle CI to kick off a build...

ktemkin, 2020-04-10:
That sounds good. The first CI build came back working -- and I think it's about time to cut a non-alpha  release of the current state. :)

By the way, @beaugunderson, would you be interested in contributor access to the project? The intention would be that you could kick off builds even they're changed to gated, and push minor fixes (e.g. URL swaps) without having to PR. :)

beaugunderson, 2020-04-10:
That would be fantastic! :)

closed 2020-04-17 by ktemkin