ktemkin/gnuradio-for-mac-without-macports #84: "VOLK required but not found" when running build script

raziele, 2020-04-12:
When build script begins configuring gnuradio for compilation it returns the following error:

```
-- Configuring VOLK support...                             
--   VOLK submodule is not checked out.                    
--   To check out the VOLK submodule, use:                 
--     git pull --recurse-submodules=on                    
--     git submodule update --init                         
--   External VOLK disabled.                               
--   Override with -DENABLE_INTERNAL_VOLK=ON/OFF           
--                                                 
CMake Error at CMakeLists.txt:428 (message):               
  VOLK required but not found. 
```

The root cause is the fact that in the gnuradio section the fetch\unpack stages are repeated in addition to the fetch\unpack stages already done in build_and_install_cmake.
The solution would be TMO to add:
```
git pull --recurse-submodules=on     
git submodule update --init`
```
inside build_and_install_cmake function

savenas, 2020-04-13:
Hi, @raziele  

did you mean like this
```
function build_and_install_cmake() {
  set -e

  local P=${1}
  local URL=${2}
  local CKSUM=${3}
  local T=${4}
  local BRANCH=${5}
  local MVFROM=${6}
  local NAME=${7}

  export -n SHELLOPTS

  if [ "" = "${T}" ]; then
    T=${P}
  fi

  if [ -f ${TMP_DIR}/.${P}.done ]; then
    I "already installed ${P}"    
  else
    fetch "${P}" "${URL}" "${T}" "${BRANCH}" "${CKSUM}"
    unpack ${P} ${URL} "${T}" "${MVFROM}" "${NAME}"
    # Create our working directory, and build things there.
    (
      set -e
      **git pull --recurse-submodules=on
      git submodule update --init**
      rm -Rf ${TMP_DIR}/${T}-build
```

raziele, 2020-04-13:
Looking into it again - I think the right place would be inside the fetch function
[here](https://github.com/ktemkin/gnuradio-for-mac-without-macports/blob/0bdc86c442c2ded616bbf4eef1df3ce56ca4c7ce/build.sh#L260) (line 255)

volk folder is added only after we checkout from the release branch, so this would be the right place to update and init

dkozel, 2020-04-13:
GNU Radio is imminently removing the VOLK submodule. Setting up the build to fetch VOLK separately would be a better long term fix.

The explanation will be up on the GNU Radio blog probably later today, but in the meantime here's the direct link to the post in a PR.
https://github.com/gnuradio/hugo-website/blob/2aa5f2ad777135ee01354cd524a745822ced90bc/content/blog/2020-04-11-removing-volk-submodule/index.md

savenas, 2020-04-13:
Thank you guys ;)