ktemkin/gnuradio-for-mac-without-macports #87: ModuleNotFoundError

snow-hdgla, 2020-04-13:
Hello! Thank you so much for undertaking this project, it'd be a huge help to me if I can get it to work!

I'm running Catalina 10.15.2. I've installed Python 3.7 and XQuartz and downloaded the app. Clicking on the app icon and nothing happens. Opening the app contents and finding the gnuradio-companion executable in bin, clicking that I get the following:

/Applications/GNURadio.app/Contents/MacOS/usr/bin/gnuradio-companion ; exit;
ModuleNotFoundError

Cannot import gnuradio.

Is the model path environment variable set correctly?
    All OS: PYTHONPATH

Is the library path environment variable set correctly?
    Linux: LD_LIBRARY_PATH
    Windows: PATH
    MacOSX: DYLD_LIBRARY_PATH

(No module named 'gnuradio')
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[Process completed]

When I try echo $PYTHONPATH and $DYLD_LIBRARY_PATH I get nothing in return. It seems I just need to set the paths appropriately, but being a novice at this kind of thing I thought I would ask what would be correct for this install. 

And, hopefully a quick second questions, if I wanted to import additional python libraries to be used inside of custom python blocks within a flow graph, would I just need to install that package into /Applications/GNURadio.app/Contents/MacOS/usr/lib/python3.7/site-packages ? 

Thank you very much for the help!

beaugunderson, 2020-04-13:
From a terminal can you execute the following command and respond with the output here?

```
/Applications/GNURadio.app/Contents/MacOS/usr/bin/run-grc
```

snow-hdgla, 2020-04-13:
Definitely! This is the output:

Scotts-MacBook-Pro:~ scottsnow$ /Applications/GNURadio.app/Contents/MacOS/usr/bin/run-grc
WARNING:gnuradio.grc.core.platform:Slow YAML loading (libyaml not available)
Traceback (most recent call last):
  File "/Applications/GNURadio.app/Contents/MacOS/usr/bin/gnuradio-companion", line 102, in <module>
    run_main()
  File "/Applications/GNURadio.app/Contents/MacOS/usr/bin/gnuradio-companion", line 95, in run_main
    exit(main())
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/gnuradio/grc/main.py", line 83, in main
    platform.build_library()
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/gnuradio/grc/core/platform.py", line 197, in build_library
    utils.hide_bokeh_gui_options_if_not_installed(self.blocks['options'])
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/collections/__init__.py", line 916, in __getitem__
    return self.__missing__(key)            # support subclasses that define __missing__
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/collections/__init__.py", line 908, in __missing__
    raise KeyError(key)
KeyError: 'options'

beaugunderson, 2020-04-13:
Oh, I encountered this as well--gnuradio 3.8 is trying to open the last file you had open when you ran gnuradio 3.7, and it's failing because the syntax is different.

This worked for me:

```
$ mv ~/.gnuradio ~/.gnuradio-old
```

You could also try to find the specific key that's responsible for specifying the file to load; I opted for starting over. :)

snow-hdgla, 2020-04-13:
Thank you so much for the help! For whatever reason I had to run 

`rm -rf ~/.gnuradio-old` 

as well, but after that it's working great! I really appreciate it!

closed 2020-04-13 by snow-hdgla