ktemkin/gnuradio-for-mac-without-macports #90: Can't import gnuradio outside of GRC

raziele, 2020-04-16:
So this is just a quick note on how to use gnuradio outside of GRC when you have other Python entities installed (By Homebrew in my case).
For example, let's sat I worked on a GRC flowgraph and now I want to edit and run its python-generated script directly (outside of GRC)
All you need to do is create a shell script with the following lines:

```
#!/bin/sh                                                                                                                                                                                                                                    
PYTHON_FRAMEWORK_DIR="/Library/Frameworks/Python.framework/Versions/${PYTHON_VERSION}"                                                                                                                                                      
PYTHON="${PYTHON_FRAMEWORK_DIR}/Resources/Python.app/Contents/MacOS/Python"                                                                                                                                                                 
source /Applications/GNURadio.app/Contents/MacOS/usr/bin/grenv.sh                                                                                                                                                                           
python $1  
```

chmod +x and then you can execute ./run_script grc_generated_script.py

Should I add this to the README \ WIKI?

beaugunderson, 2020-04-16:
I've been wanting to figure out the best way to launch a shell with the
right environment, this is one approach... In my ideal scenario we'd come
up with a `gr-shell` command which launches a shell with the environment
applied, how do you feel about that?

raziele, 2020-04-16:
That's also an option. I prefer not to launch a new shell as I use tmux and prefer maintaining the same environment across all windows and panes, but perhaps there's a cool way to do it that i'm missing.
Anyway, why not showing both options or more? the principle is the same: source grenv.sh along with the beginning of run-grc and then do whatever

beaugunderson, 2020-04-16:
agreed, showing both ways would be the most flexible 👍

ricky-bos, 2020-04-22:
I still cannot run the python-script :'( 

Got the following message in terminal 

> Traceback (most recent call last):
  File "top_block.py", line 24, in <module>
    from gnuradio import qtgui
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/gnuradio/qtgui/__init__.py", line 38, in <module>
    from .range import Range, RangeWidget
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/gnuradio/qtgui/range.py", line 25, in <module>
    from .util import check_set_qss
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/gnuradio/qtgui/util.py", line 28, in <module>
    from gnuradio import gr
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/gnuradio/gr/__init__.py", line 46, in <module>
    from .top_block import *
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/gnuradio/gr/top_block.py", line 32, in <module>
    from .hier_block2 import hier_block2
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/gnuradio/gr/hier_block2.py", line 26, in <module>
    import pmt
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/pmt/__init__.py", line 61, in <module>
    from .pmt_to_python import pmt_to_python as to_python
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/pmt/pmt_to_python.py", line 23, in <module>
    import numpy
  File "/usr/local/lib/python3.7/site-packages/numpy/__init__.py", line 152, in <module>
    from . import random
  File "/usr/local/lib/python3.7/site-packages/numpy/random/__init__.py", line 181, in <module>
    from . import _pickle
  File "/usr/local/lib/python3.7/site-packages/numpy/random/_pickle.py", line 1, in <module>
    from .mtrand import RandomState
  File "mtrand.pyx", line 1, in init numpy.random.mtrand
ModuleNotFoundError: No module named 'numpy.random.common'

beaugunderson, 2020-04-22:
@ricky-bos can you create a new issue for that failure? (I can tell you that you'll want to install `numpy`, but if you create a new issue I'm happy to help troubleshoot further)