ktemkin/gnuradio-for-mac-without-macports #91: Cannot Save or run flow graphs

adamtdiscog, 2020-05-02:
When I try to build the flow graph and save the grc file, it returns with 

"Block paths:
	/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/grc/blocks
>>> Error: Cannot save: /test.grc"

Trying to save in a different file location also return an error:

"Could not read the contents of Desktop" "Error opening directory, operation not permitted."

joelaso, 2020-05-19:
Hi @adamtdiscog, if you run `sudo /Applications/GNURadio.app/Contents/MacOS/usr/bin/run-grc` in terminal, it runs gun radio companion in sudo mode and you should be able to access directories you couldn't before.

adamtdiscog, 2020-05-20:
@joelaso Thanks! Should have known to try something like that....always the simple stuff that gets me.

closed 2020-05-20 by adamtdiscog