ktemkin/gnuradio-for-mac-without-macports #92: Can't run a diagram with the osmocom source

tomszr-hid, 2020-05-08:
Stright after installation GNU Radio is working but I'm not able to start any diagram with the osmocom Source. I'm getting a trace:
"<<< Welcome to GNU Radio Companion 3.8.0.0 >>>

Block paths:
	/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/grc/blocks

Loading: "/Users/tomek/RadioFM.grc"
>>> Done

Loading: "/Users/tomek/test.grc"
>>> Done

Generating: '/Users/tomek/some_test.py'
>>> Warning: This flow graph may not have flow control: no audio or RF hardware blocks found. Add a Misc->Throttle block to your flow graph to avoid CPU congestion.

Executing: /Library/Frameworks/Python.framework/Versions/3.7/Resources/Python.app/Contents/MacOS/Python -u /Users/tomek/some_test.py

>>> Done

Generating: '/Users/tomek/top_block.py'

Executing: /Library/Frameworks/Python.framework/Versions/3.7/Resources/Python.app/Contents/MacOS/Python -u /Users/tomek/top_block.py

Traceback (most recent call last):
  File "/Users/tomek/top_block.py", line 38, in <module>
    import osmosdr
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/osmosdr/__init__.py", line 25, in <module>
    from .osmosdr_swig import *
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/osmosdr/osmosdr_swig.py", line 13, in <module>
    from . import _osmosdr_swig
ImportError: dlopen(/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/osmosdr/_osmosdr_swig.so, 2): Library not loaded: /usr/local/opt/python/Frameworks/Python.framework/Versions/3.7/Python
  Referenced from: /Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/osmosdr/_osmosdr_swig.so
  Reason: image not found

>>> Done (return code 1)
"

iivanu, 2020-06-23:
I have same problem, hope someone will find solution.

RocketManRC, 2020-11-04:
This is my first time experimenting with GNURadio and I ran into this problem as well. I believe that the problem is that _osmosdr_swig.so references Python twice and the path to the second reference doesn't exist. That can be seen with the following command:

$ otool -L /Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/osmosdr/_osmosdr_swig.so

which gives:

/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/osmosdr/_osmosdr_swig.so:
	/Library/Frameworks/Python.framework/Versions/3.7/Python (compatibility version 3.7.0, current version 3.7.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1281.0.0)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1670.10.0)
	/Applications/GNURadio.app/Contents/MacOS/usr/lib/libgnuradio-osmosdr.0.2.0.dylib (compatibility version 0.2.0, current version 0.2.0)
	/usr/local/opt/python/Frameworks/Python.framework/Versions/3.7/Python (compatibility version 3.7.0, current version 3.7.0)
	...

I was able to fix the problem by running this command which changes the second Python reference in _osmosdr_swig.so to the correct one:

$ install_name_tool -change /usr/local/opt/python/Frameworks/Python.framework/Versions/3.7/Python  /Library/Frameworks/Python.framework/Versions/3.7/Python /Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/osmosdr/_osmosdr_swig.so

I didn't try to look into why the library has that second reference but I suspect it has something to do with Python 3.7 having been installed on the build machine via brew at some point.

m1geo, 2021-02-15:
Thanks @RocketManRC - This worked a treat. Down side is that it needs to install Xcode; but, it did get things working :)

johnnyhuziqin, 2021-08-01:
I also have this issue using my hack rf one, the core reason is there is no /usr/local/opt/python/Frameworks/Python.framework/Versions/3.7/Python exist, but I found that there is python 3.9 in the same path, so I make a soft link to that python dir, and it turn out working.
`$ cd /usr/local/opt/python/Frameworks/Python.framework/Versions
ln -s 3.9 3.7
`

kevbost, 2021-12-22:
> I was able to fix the problem by running this command which changes the second Python reference in _osmosdr_swig.so to the correct one:
>
> $ install_name_tool -change [...]

I tried a lot of things to fix this same issue, but @RocketManRC's install_name_tool suggestion finally got my RTL-SDR working for me on MacOS. I saw that it is mentioned on the official MacInstall section of the gnuradio wiki, but just don't know enough about these systems yet to have known that was the solution.

https://wiki.gnuradio.org/index.php/MacInstall
Search page for `install_name_tool`

Following the simplest bare-bones tutorial I've seen on the topic:
https://www.youtube.com/watch?v=aG1j6Fyyc8A