ktemkin/gnuradio-for-mac-without-macports #93: build.sh errors with CppUnit

zendata, 2020-05-09:
Tried building on two different MBP's, one running Catalina and the other Mojave and both stop the build script with the same error(s):

Making install in DllPlugInTester
clang++ -mmacosx-version-min=10.7 -stdlib=libc++ -DHAVE_CONFIG_H -I. -I../../config -I../../include -I../../include  -I/Applications/GNURadio.app/Contents/MacOS/usr/include -I/opt/X11/include  -I/Applications/GNURadio.app/Contents/MacOS/usr/include -I/opt/X11/include -MT DllPlugInTester.o -MD -MP -MF .deps/DllPlugInTester.Tpo -c -o DllPlugInTester.o DllPlugInTester.cpp
clang++ -mmacosx-version-min=10.7 -stdlib=libc++ -DHAVE_CONFIG_H -I. -I../../config -I../../include -I../../include  -I/Applications/GNURadio.app/Contents/MacOS/usr/include -I/opt/X11/include  -I/Applications/GNURadio.app/Contents/MacOS/usr/include -I/opt/X11/include -MT CommandLineParser.o -MD -MP -MF .deps/CommandLineParser.Tpo -c -o CommandLineParser.o CommandLineParser.cpp
In file included from CommandLineParser.cpp:1:
./CommandLineParser.h:26:3: error: use of undeclared identifier 'CppUnit'
  CPPUNIT_NS::PlugInParameters m_parameters;
  ^
../../include/cppunit/Portability.h:121:21: note: expanded from macro 'CPPUNIT_NS'
# define CPPUNIT_NS CppUnit
                    ^
CommandLineParser.cpp:78:29: error: use of undeclared identifier 'CppUnit'
      plugIn.m_parameters = CPPUNIT_NS::PlugInParameters( parameters );
                            ^
../../include/cppunit/Portability.h:121:21: note: expanded from macro 'CPPUNIT_NS'
# define CPPUNIT_NS CppUnit
                    ^
2 errors generated.
make[2]: *** [CommandLineParser.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from DllPlugInTester.cpp:16:
./CommandLineParser.h:26:15: error: no type named 'PlugInParameters' in namespace 'CppUnit'
  CPPUNIT_NS::PlugInParameters m_parameters;
  ~~~~~~~~~~~~^
DllPlugInTester.cpp:35:15: error: no type named 'PlugInManager' in namespace 'CppUnit'
  CPPUNIT_NS::PlugInManager plugInManager;
  ~~~~~~~~~~~~^
DllPlugInTester.cpp:240:23: error: no type named 'DynamicLibraryManagerException' in namespace 'CppUnit'
  catch ( CPPUNIT_NS::DynamicLibraryManagerException &e )
          ~~~~~~~~~~~~^
3 errors generated.
make[2]: *** [DllPlugInTester.o] Error 1
make[1]: *** [install-recursive] Error 1
make: *** [install-recursive] Error 1

jhegedus42, 2020-06-01:
similar problem here : 

```
Making install in DllPlugInTester
clang++ -mmacosx-version-min=10.7 -stdlib=libc++ -DHAVE_CONFIG_H -I. -I../../config -I../../include -I../../include  -I/Applications/GNURadio.app/Contents/MacOS/usr/include -I/opt/X11/include  -I/Applications/GNURadio.app/Contents/MacOS/usr/include -I/opt/X11/include -MT DllPlugInTester.o -MD -MP -MF .deps/DllPlugInTester.Tpo -c -o DllPlugInTester.o DllPlugInTester.cpp
clang++ -mmacosx-version-min=10.7 -stdlib=libc++ -DHAVE_CONFIG_H -I. -I../../config -I../../include -I../../include  -I/Applications/GNURadio.app/Contents/MacOS/usr/include -I/opt/X11/include  -I/Applications/GNURadio.app/Contents/MacOS/usr/include -I/opt/X11/include -MT CommandLineParser.o -MD -MP -MF .deps/CommandLineParser.Tpo -c -o CommandLineParser.o CommandLineParser.cpp
In file included from CommandLineParser.cpp:1:
./CommandLineParser.h:26:3: error: use of undeclared identifier 'CppUnit'
  CPPUNIT_NS::PlugInParameters m_parameters;
  ^
../../include/cppunit/Portability.h:121:21: note: expanded from macro 'CPPUNIT_NS'
# define CPPUNIT_NS CppUnit
                    ^
CommandLineParser.cpp:78:29: error: use of undeclared identifier 'CppUnit'
      plugIn.m_parameters = CPPUNIT_NS::PlugInParameters( parameters );
                            ^
../../include/cppunit/Portability.h:121:21: note: expanded from macro 'CPPUNIT_NS'
# define CPPUNIT_NS CppUnit
                    ^
2 errors generated.
make[2]: *** [CommandLineParser.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from DllPlugInTester.cpp:16:
./CommandLineParser.h:26:15: error: no type named 'PlugInParameters' in namespace 'CppUnit'
  CPPUNIT_NS::PlugInParameters m_parameters;
  ~~~~~~~~~~~~^
DllPlugInTester.cpp:35:15: error: no type named 'PlugInManager' in namespace 'CppUnit'
  CPPUNIT_NS::PlugInManager plugInManager;
  ~~~~~~~~~~~~^
DllPlugInTester.cpp:240:23: error: no type named 'DynamicLibraryManagerException' in namespace 'CppUnit'
  catch ( CPPUNIT_NS::DynamicLibraryManagerException &e )
          ~~~~~~~~~~~~^
3 errors generated.
make[2]: *** [DllPlugInTester.o] Error 1
make[1]: *** [install-recursive] Error 1
make: *** [install-recursive] Error 1
E: failed to configure, make, and install cppunit-1.12.1
```

jhegedus42, 2020-06-01:
[image]
This is the op system ^

jhegedus42, 2020-06-01:
This is the op system : 

[image]

markl17, 2020-06-04:
hi similar prob here and it is Catalina 10.15.6 
./CommandLineParser.h:26:15: error: no type named
      'PlugInParameters' in namespace 'CppUnit'
  CPPUNIT_NS::PlugInParameters m_parameters;
  ~~~~~~~~~~~~^
DllPlugInTester.cpp:35:15: error: no type named
      'PlugInManager' in namespace 'CppUnit'
  CPPUNIT_NS::PlugInManager plugInManager;
  ~~~~~~~~~~~~^
DllPlugInTester.cpp:240:23: error: no type named
      'DynamicLibraryManagerException' in namespace
      'CppUnit'
  catch ( CPPUNIT_NS::DynamicLibraryManagerException &e )
          ~~~~~~~~~~~~^

zendata, 2020-07-07:
Hi @markl17 and @jhegedus42 - did either of you ever resolve our mutual problem?

jfamestad, 2020-07-14:
I am also hitting this issue. @markl17 @jhegedus42 @zendata - Have any of you found a resolution?

zendata, 2020-07-14:
No unfortunately. I have tried everything I can think of and retested last week - still getting the same error.

Cheers,
Graham Cottew
Technology Director
Zendata Pty Ltd
+61417722400

raziele, 2020-12-05:
I managed to bypass this build step by doing the following steps:
After the first error, I entered the cppunit-1.12.1 folder under ./tmp and compiled manually:
```
./configure --prefix=/Applications/GNURadio.app/Contents/MacOS/usr
make
make install
```

then I manually signalled the build script this step is done by:
```
touch cppunit-1.12.1.done
```

and rerun build.sh