ktemkin/gnuradio-for-mac-without-macports #94: Failed to open gnuradio companion on 10.15.4

joelaso, 2020-05-16:
Hello, 

I installed gnuradio via the dmg install and it will not open.  I see scrip notifications for openginfng X11 and executing companion, but nothing happens.  this is what I see in terminal.

/Applications/GNURadio.app/Contents/MacOS/usr/bin/run-grc ; exit;
WARNING:gnuradio.grc.core.platform:Slow YAML loading (libyaml not available)
Traceback (most recent call last):
  File "/Applications/GNURadio.app/Contents/MacOS/usr/bin/gnuradio-companion", line 102, in <module>
    run_main()
  File "/Applications/GNURadio.app/Contents/MacOS/usr/bin/gnuradio-companion", line 95, in run_main
    exit(main())
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/gnuradio/grc/main.py", line 83, in main
    platform.build_library()
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/gnuradio/grc/core/platform.py", line 197, in build_library
    utils.hide_bokeh_gui_options_if_not_installed(self.blocks['options'])
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/collections/__init__.py", line 916, in __getitem__
    return self.__missing__(key)            # support subclasses that define __missing__
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/collections/__init__.py", line 908, in __missing__
    raise KeyError(key)
KeyError: 'options'
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

Thank you in advance

beaugunderson, 2020-05-16:
hey @joelaso, I believe this is the same issue as https://github.com/ktemkin/gnuradio-for-mac-without-macports/issues/75 and https://github.com/ktemkin/gnuradio-for-mac-without-macports/issues/87; can you try moving your `~/.gnuradio` to `~/.old-gnuradio` and see if the problem resolves?

joelaso, 2020-05-18:
@beaugunderson Thank you very much! Sorry for the duplicate posting.

closed 2020-05-19 by joelaso