ktemkin/gnuradio-for-mac-without-macports #96: GNU Radio 3.8 failed to start (error 127)

ah1102, 2020-05-21:
[image]

```
Last login: Thu May 21 19:32:50 on ttys000
Apple@iMac-Apple ~ % /Applications/GNURadio.app/Contents/MacOS/usr/bin/run-grc 
/Applications/GNURadio.app/Contents/MacOS/usr/bin/run-grc: /Applications/GNURadio.app/Contents/MacOS/usr/bin/gnuradio-companion: /Library/Frameworks/Python.framework/Versions/3.7/Resources/Python.app/Content: bad interpreter: No such file or directory
/Applications/GNURadio.app/Contents/MacOS/usr/bin/run-grc: line 108: /Applications/GNURadio.app/Contents/MacOS/usr/bin/gnuradio-companion: Undefined error: 0
Apple@iMac-Apple ~ % 
```

beaugunderson, 2020-05-21:
hey @ah1102, you'll want to install the python.org version of Python 3.7, available here:

https://www.python.org/downloads/release/python-377/

this is noted in the README:

> You must install Python 3.7 using the python.org installer; GNURadio.app can't use the version installed with macOS (or MacPorts or Homebrew).

the python.org should coexist peacefully with any other versions of Python you have installed. :)

ah1102, 2020-05-21:
Give thanks! Why don't you get the program to get the Python location from the environment variable?

beaugunderson, 2020-05-21:
@ah1102 I agree that it would be an improvement if we could use an already-installed version of Python; [this comment](https://github.com/ktemkin/gnuradio-for-mac-without-macports/issues/70#issuecomment-542490687) explains the reason that we can't just yet :)

closed 2020-05-21 by beaugunderson