ktemkin/gnuradio-for-mac-without-macports #100: Cannot access files

BWorp, 2020-05-31:
Hello,

I installed latest gnuradio via the dmg install and it will start gnu radio companion OK. Using latest XQuartz and Python 3.7.7. MacBook 13", 2020, 10.15.5
If I want to open an existing project in my documents folder I get the message "Error opening directory ... operation not permitted". Also opening files on external mounted disk files the same way. I assume some Apple security layer frustrates access. I do not know MacOs specific security mechanisms enough to dive into this myself.

Thanks and kind regards, Bart.

BWorp, 2020-06-01:
And, forgot to mention that before, thank you for providing this GnuRadio installation option!

nsbruce, 2020-07-08:
Same problem. Did you find a resolution @BWorp ?

docquantum, 2020-07-25:
This was discussed in other open source projects such as GIMP. The easy work around is to give /bin/sh "Full Disk Access" since this is a console application. This is not desirable however because that allows any console application full access to your system, which may mean malicious code can ruin things on your system.

This is discussed and within [this issue](https://gitlab.gnome.org/GNOME/gimp/-/issues/3710) on GIMP's git. One of the devs had [fixed the issue](https://gitlab.gnome.org/GNOME/gimp/-/issues/3710#note_649970) by moving the shell script logic into the gimp executable so that the appbundle's entrypoint was the executable, rather than a shell script wrapper. Something similar may need to happen here. Or maybe there's a way to do this with AppleScript.

The correct fix would be to implement the security features as required by apple which can be found [here](https://developer.apple.com/documentation/bundleresources/information_property_list/protected_resources). I'm not sure how to go about this however as I am not an Apple developer.