ktemkin/gnuradio-for-mac-without-macports #101: Reason: image not found

iivanu, 2020-06-23:
I've followed steps and when I want to use it with HackRF One, there is error while compiling.

Executing: /Library/Frameworks/Python.framework/Versions/3.7/Resources/Python.app/Contents/MacOS/Python -u /Users/ivanivanusic/defaultfsdaf.py

Traceback (most recent call last):
  File "/Users/ivanivanusic/defaultfsdaf.py", line 33, in <module>
    import osmosdr
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/osmosdr/__init__.py", line 25, in <module>
    from .osmosdr_swig import *
  File "/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/osmosdr/osmosdr_swig.py", line 13, in <module>
    from . import _osmosdr_swig
ImportError: dlopen(/Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/osmosdr/_osmosdr_swig.so, 2): Library not loaded: /usr/local/opt/python/Frameworks/Python.framework/Versions/3.7/Python
  Referenced from: /Applications/GNURadio.app/Contents/MacOS/usr/share/gnuradio/python/site-packages/osmosdr/_osmosdr_swig.so
  Reason: image not found

>>> Done (return code 1)

cloudn1ne, 2020-07-11:
same for me

BG7JAF, 2020-08-20:
same here with rtlsdr

iivanu, 2020-10-07:
didnt fixed this

closed 2020-10-07 by iivanu