ktemkin/gnuradio-for-mac-without-macports #103: Problem trying to run gr_modtool

tonytd48, 2020-07-12:
When I run the gr_modtool script this is what happens;

tonytd@MacBook-Pro ~ % gr_modtool -d ./Physical_Layer
Traceback (most recent call last):
  File "/Applications/GNURadio.app/Contents/MacOS/usr/bin/gr_modtool", line 25, in <module>
    from gnuradio.modtool.cli.base import cli
ModuleNotFoundError: No module named 'gnuradio.modtool'

If I run Python3.7 and import the module "gnuradio", then I find that modtool does not exist within it, which might explain the above. The environment is set up as per grenv.sh in both cases.

Is there some other way of running gr_modtool? What am I missing?

Running Catalina 10.15.5 and Xcode 11.5

raziele, 2020-12-05:
I experience the same behavior, Xcode 12.2

luigifcruz, 2021-07-14:
I think this is probably related to a couple of missing dependencies `python3-click python3-click-plugins`.