ktemkin/gnuradio-for-mac-without-macports #109: Doesn't start...

RolfNoot, 2020-11-06:
After it shows: 
- Starting X server
- Launching GNU Radio Companion
nothing happens.

I've installed Python and XQuartz.

-Rolf

RolfNoot, 2020-11-06:
Seems the location of Python is not okay:
-bash: /Applications/GNURadio/usr/bin/gnuradio-companion: /Library/Frameworks/Python.framework/Versions/3.7/Resources/Python.app/Content: bad interpreter: No such file or directory

How can I change the '3.7' into 'Current'?

-Rolf

RolfNoot, 2020-11-06:
Installing Python 3.7 did the trick. Maybe change the Python link inside the read.me to the 3.7 version: 
https://www.python.org/downloads/release/python-377/

m1geo, 2021-02-15:
Also had this issue. I too looked and found the Python script issue. Changing to the Python version linked by @RolfNoot got it working. It's slow to launch but works OK (Mac Air M1 though Rosetta).

ThermoNuclear65, 2021-07-21:
Yes .Also had this issue —>solved .Thanks to RolfNoot !