ktemkin/gnuradio-for-mac-without-macports #118: Have you thought about making a homebrew cask?

zellyn, 2021-08-17:
It would be lovely to have a homebrew cask that can install this.

marcopovitch, 2022-04-17:
there is one (currently version 3.9.3.0_4) :

`brew install gnuradio`

more info at : [https://wiki.gnuradio.org/index.php/InstallingGR](https://wiki.gnuradio.org/index.php/InstallingGR)

zebrapurring, 2024-05-31:
Here's my current approach to create an application based on the Homebrew installation:

```sh
#!/bin/sh

set -eu

app_path="${1:-"/Applications/GNURadio.app"}"

FORMULA_NAME="gnuradio"

# Create application directories
mkdir -p "$app_path/Contents/MacOS"
mkdir -p "$app_path/Contents/Resources"

# Download icons
curl -s -o "$app_path/Contents/Resources/gnuradio.icns" "https://raw.githubusercontent.com/ktemkin/gnuradio-for-mac-without-macports/560d51574f876f8a44829ba416737dfa965953b5/gnuradio.icns"

# Link current Homebrew installation
if ! brew list --formula "$FORMULA_NAME" > /dev/null 2>&1; then
    echo "GNURadio installation not found, please install with Homebrew first:" >&2
    echo "$ brew install $FORMULA_NAME" >&2
    exit 1
fi
installed_version="$(brew list --formulae --versions | grep "^$FORMULA_NAME " | cut -d " " -f 2 | cut -d "_" -f 1)"
ln -sf "$(brew --prefix "$FORMULA_NAME")/bin/gnuradio-companion" "$app_path/Contents/MacOS"

# Create application manifest
cat > "$app_path/Contents/Info.plist" << EOF
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple Computer//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
  <key>CFBundleGetInfoString</key>
  <string>GNURadio</string>
  <key>CFBundleExecutable</key>
  <string>gnuradio-companion</string>
  <key>CFBundleIdentifier</key>
  <string>org.gnuradio.gnuradio-companion</string>
  <key>CFBundleName</key>
  <string>GNURadio</string>
  <key>CFBundleIconFile</key>
  <string>gnuradio.icns</string>
  <key>CFBundleShortVersionString</key>
  <string>$installed_version</string>
  <key>CFBundleInfoDictionaryVersion</key>
  <string>6.0</string>
  <key>CFBundlePackageType</key>
  <string>APPL</string>
  <key>CFBundleDocumentTypes</key>
  <array>
    <dict>
      <key>CFBundleTypeExtensions</key>
      <array>
        <string>grc</string>
        <string>GRC</string>
        <string>grc.xml</string>
        <string>GRC.XML</string>
      </array>
      <key>CFBundleTypeIconFile</key>
      <string>gnuradio.icns</string>
      <key>CFBundleTypeMIMETypes</key>
      <array>
        <string>application/gnuradio-grc</string>
      </array>
      <key>CFBundleTypeName</key>
      <string>GNU Radio Companion Flow Graph</string>
      <key>CFBundleTypeOSTypes</key>
      <array>
        <string>GRC </string>
      </array>
      <key>CFBundleTypeRole</key>
      <string>Editor</string>
      <key>LSIsAppleDefaultForType</key>
      <true />
      <key>LSItemContentTypes</key>
      <array>
        <string>org.gnuradio.grc</string>
      </array>
    </dict>
  </array>
  <key>UTExportedTypeDeclarations</key>
  <array>
    <dict>
      <key>UTTypeConformsTo</key>
      <array>
        <string>public.xml</string>
      </array>
      <key>UTTypeDescription</key>
      <string>GNU Radio Companion Flow Graph</string>
      <key>UTTypeIconFile</key>
      <string>gnuradio.icns</string>
      <key>UTTypeIdentifier</key>
      <string>org.gnuradio.grc</string>
      <key>UTTypeReferenceURL</key>
      <string>http://www.gnuradio.org/</string>
      <key>UTTypeTagSpecification</key>
      <dict>
        <key>com.apple.ostype</key>
        <string>GRC </string>
        <key>public.filename-extension</key>
        <array>
          <string>grc</string>
          <string>GRC</string>
          <string>grc.xml</string>
          <string>GRC.XML</string>
        </array>
        <key>public.mime-type</key>
        <array>
          <string>application/gnuradio-grc</string>
        </array>
      </dict>
    </dict>
  </array>
</dict>
</plist>
EOF

echo "Done! You can now open the new application at $app_path"
```

If you upgrade GNURadio, you just need to re-run the script to update the version in the manifest.