ktemkin/gnuradio-for-mac-without-macports #120: Doesn't build on M1/M2 Macs

JKI757, 2022-07-07:
The release doesn't run on M1 Macs, it seems to start x-windows and then just quits, doesn't give any other indication of what's wrong.  So trying to build using these scripts (after fixing the boost download issue and the lib tool download issue and manually installing python 3.9 and adjusting the build script to point to that install), it runs into trouble compiling ffi -- seems to be an arm assembly issue perhaps.  This one is a bit over my head.  Maybe someone has an idea on how to resolve it?  I've attached my modified build script in case I made a mistake there that's causing the issue.

[build.sh.txt](https://github.com/ktemkin/gnuradio-for-mac-without-macports/files/9064736/build.sh.txt)

partial compile output:

libtool: compile:  clang -mmacosx-version-min=10.7 -DHAVE_CONFIG_H -I. -I.. -I. -I../include -Iinclude -I../src -I/Applications/GNURadio.app/Contents/MacOS/usr/include -I/opt/X11/include -I/Applications/GNURadio.app/Contents/MacOS/usr/include -I/opt/X11/include -Wall -fexceptions -MT src/arm/ffi.lo -MD -MP -MF src/arm/.deps/ffi.Tpo -c ../src/arm/ffi.c  -fno-common -DPIC -o src/arm/.libs/ffi.o
../src/arm/sysv.S:64:1: error: unexpected token at start of statement
@ This selects the minimum architecture level required.
^
../src/arm/sysv.S:112:1: error: unknown directive
.syntax unified
^
../src/arm/sysv.S:152:2: error: unexpected token at start of statement
 @ r0: ffi_prep_args
 ^
../src/arm/sysv.S:153:2: error: unexpected token at start of statement
 @ r1: &ecif
 ^
../src/arm/sysv.S:154:2: error: unexpected token at start of statement
 @ r2: cif->bytes

 ^
../src/arm/sysv.S:155:2: error: unexpected token at start of statement
 @ r3: fig->flags
 ^

JKI757, 2022-07-07:
Ok, fixed the libffi problem using this link:  https://github.com/libffi/libffi/issues/571

adding the parameter "--build=aarch64-apple-darwin20.0.0" to CONFIGURE_CMD on line 556 of the build script allows libffi to build correctly.  Apparently it detects the architecture incorrectly in libffi's configure script and this forces it.  

I'm going to just go ahead and document my modifications to the build script as I go along trying to get this to work, hopefully can come up with a script that works on apple silicon eventually.

dmaltsiniotis, 2022-10-04:
> Ok, fixed the libffi problem using this link: [libffi/libffi#571](https://github.com/libffi/libffi/issues/571)
> 
> adding the parameter "--build=aarch64-apple-darwin20.0.0" to CONFIGURE_CMD on line 556 of the build script allows libffi to build correctly. Apparently it detects the architecture incorrectly in libffi's configure script and this forces it.
> 
> I'm going to just go ahead and document my modifications to the build script as I go along trying to get this to work, hopefully can come up with a script that works on apple silicon eventually.

Thanks for following up and posting your work on this, it's very helpful.

krono86, 2022-10-09:
+1 I’m really interested in compiling gnuradio for Apple silicon.
Please, keep me updated!

JKI757, 2022-10-21:
You're missing the meson package. Most likely it's just not installed in such a way as to be accessible to that version of python.

JKI757, 2022-10-21:
I'm not at a computer right now, but even though meson is installed it's apparently not available to the python version you're using. 

The basic gist is that you need to run python pip3 -m install meson under the version of python you are using. In other words you're going to need to run that command, but using the full path of the python version that is being used to build this. 

Python on Mac is such a pain to work with...

krono86, 2022-10-24:
I'm installing meson through the script itself.
I found out meson has been correctly installed in `INSTALL_PYTHON_DIR` (i.e., `/Applications/GNURadio.app/Contents/MacOS/usr/lib/python3.9/site-packages`); there is `meson-0.63.3-py3.9.egg` and it is linked in `easy-install.pth`.
The problem is that when meson is called, python does not find it because the egg file is not in `sys.path`.
Maybe it is a problem in the meson installation. Find attached the relevant lines of the build script.
```
function build_and_install_setup_py() {
  set -e

  local P=${1}
  local URL=${2}
  local CKSUM=${3}
  local T=${4}
  local BRANCH=${5}

  export -n SHELLOPTS

  if [ "" = "${T}" ]; then
    T=${P}
  fi

  if [ -f ${TMP_DIR}/.${P}.done ]; then
    I "already installed ${P}"    
  else 
  
    fetch "${P}" "${URL}" "${T}" "${BRANCH}" "${CKSUM}"
    unpack ${P} ${URL} ${T}
  
    if [ ! -d ${PYTHONPATH} ]; then
      mkdir -p ${PYTHONPATH} \
        || E "failed to mkdir -p ${PYTHONPATH}"
    fi
  
    I "Configuring and building in ${T}"
    cd ${TMP_DIR}/${T} \
      && \
        ${PYTHON} setup.py install --prefix=${INSTALL_DIR}/usr \
      || E "failed to configure and install ${P}"
  
    I "finished building and installing ${P}"
    touch ${TMP_DIR}/.${P}.done
    
  fi
}

...

#
# Install meson
# 
(
  V=0.63.3
  P=meson-${V}
  URL="https://github.com/mesonbuild/meson/releases/download/${V}/${P}.tar.gz"
  CKSUM=sha256:519c0932e1a8b208741f0fdce90aa5c0b528dd297cf337009bf63539846ac056

  build_and_install_setup_py \
    ${P} \
    ${URL} \
    ${CKSUM}
)
```